sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #5: UBX file not recognized in RTKLIB

johnhulth, 2021-01-16:
Hi, I'm trying [DataLoggingExample3_RXM_SFRBX_and_RAWX](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Data_Logging/DataLoggingExample3_RXM_SFRBX_and_RAWX/DataLoggingExample3_RXM_SFRBX_and_RAWX.ino) to log a UBX-file. But the file is not recognized by RTKCONV and i can not open it in u-center. The file size is 2.3MB, so there is data on it. I have tried several times.

I'm using:
Feather Adalogger M0 and connect to ZED-F9P with the Qwiic connector
RTKCONV ver.2.4.2
u-Center 20.10

PaulZC, 2021-01-16:
Hi John,
Do you have Python 3 available? If you do, there is a utility [here](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/master/Utils) which can check the integrity of the file. Please post the results here.
Or, if you are happy for me to see your data, you can email it to me and I will take a look. You can find my email address on my profile page (just click on my "PC" avatar). I will of course delete it afterwards.
Best wishes,
Paul

johnhulth, 2021-01-16:
Here is the result: 

UBX Sync Char 1 (0xB5) found at byte 0. Checking for Sync Char 2

```
Processed 2433376 bytes
File size was 2433376
Longest UBX message was 1616 data bytes
Message types and totals were:
Message type: 0x02 0x15   Total: 1135
Message type: 0x02 0x13   Total: 16193
```
Looks good to me, but when i try to convert to RINEX with RTKCONV or open the file in u-Ceneter, there is nothing.

PaulZC, 2021-01-16:
Resolved via email.... (With the correct settings, both RTKCONV and u-center can open the file.)

closed 2021-01-16 by PaulZC

slowerNSteady, 2021-01-30:
Hi All,
I have a similar issue so decided to post here rather than start a new issue.
I'm using the data logging example number 3, but trying to modify it to suit using the Redboard Artemis Nano  [DEV-15443] with a Qwiic Openlog [ DEV-15164 ].
I can get a file created on the sd Card that looks similar to what u-centre produces but it won't convert to a rinex file when using RTKCONV.
Heres my code:
[d.txt](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/files/5897688/d.txt)

I'm pretty new to programming and have only studied java script, so not very competent with microcontrollers etc.

Any help would be much appreciated as this is for a uni project I will be starting in a few weeks. If it's too hard I plan on purchasing the serial version of the openlog.
Cheers
Josh

PaulZC, 2021-01-30:
Hi Josh (@slowerNSteady ),
Your code looks like it should work OK, but I don't have an OpenLog and so I can't replicate what you are seeing... Running both the GNSS and the OpenLog on the same I2C bus _should_ be OK, especially if you are logging slowly (at 0.5Hz)... I suspect you are dropping data, but I can't tell...
If you have Python (3), try running your log file through the [UBX Integrity Checker](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/master/Utils). That will tell you if the data is complete or if it is dropping out.
I see you are writing 32 bytes at a time. That's good. I was going to suggest changing sdWriteSize, but you've done that.
Pull-ups could be an issue. The GNSS works best without pull-ups, but the OpenLog will need them. You can try uncommenting the ```myGNSS.enableDebugging(Serial, true);```. The debug messages will tell you if the I2C data is being corrupted. But only clean data which passes the checksum test gets added to the file buffer, so that's not likely to be the main issue.
I was going to suggest using a second I2C Wire port for the OpenLog, but I see the Nano only has one Wire interface defined.
If you want to, you can email your log file to me and I will take a quick look. You can find my email on my GitHub Profile.
Best wishes,
Paul

PaulZC, 2021-01-30:
Also, maybe delete the ```myLog.append("RXM_RAWX.ubx");``` in the main loop. It looks like that is redundant - and might cause problems?
Also, maybe delete the ```myLog.syncFile();``` after the first myLog.write for loop. So long as you do a sync when you 'close' the file, you should be OK.