sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #6: getPositionAccuracy() not returning correct values ?

Niv1900, 2021-01-19:
Hi @PaulZC,

Thank you for this 2.0.x version. I'm quite a "baby user" here since it's my first issue that I post on github... 

I'm having some trouble to use the _getPositionAccuracy()_ function. I'll try to not miss any piece of information. 

### Workbench

- NEO-M8N (this product is not [officially marked as compatible](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library#products-that-use-this-library), but so far all functions are working correctly) ;
- ESP32 (lolin32) with the Arduino framework ;
- I²C connection without addition pull-up resistor ;
- I²C tested at 400 000 Hz and 40 000 Hz ;
- Everything is powered via the USB cable on the ESP32 ;
- an IMU is running in parallel via SPI Bus ;
- compiled code with -O3 flag.

### Steps to reproduce
Ok so I'm quite a begginer here, so what I do is that I get all the info I need and put them into a struct for a use in another task, here is my GNSS Task : 

```
void xT_GNSS(void *pvParameters)
{
    TickType_t xLastWakeTime_GNSS;
    xLastWakeTime_GNSS = xTaskGetTickCount();

    while (1)
    {
        vTaskDelayUntil(&xLastWakeTime_GNSS, Tck_GNSS);

        F.GNSS.fix           = myGNSS.getGnssFixOk();
        F.GNSS.latitude      = myGNSS.getLatitude();
        F.GNSS.longitude     = myGNSS.getLongitude();
        F.GNSS.altitude      = myGNSS.getAltitude();
        F.GNSS.fixType       = myGNSS.getFixType();
        F.GNSS.groundSpeed   = myGNSS.getGroundSpeed();
        F.GNSS.heading       = myGNSS.getHeading();
        F.GNSS.PDOP          = myGNSS.getPDOP();
        F.GNSS.headingVeh    = myGNSS.getHeadVeh();
        F.GNSS.SIV           = myGNSS.getSIV();
        F.GNSS.year          = myGNSS.getYear();
        F.GNSS.month         = myGNSS.getMonth();
        F.GNSS.day           = myGNSS.getDay();
        F.GNSS.hour          = myGNSS.getHour();
        F.GNSS.minute        = myGNSS.getMinute();
        F.GNSS.second        = myGNSS.getSecond();
        F.GNSS.millisecond   = myGNSS.getMillisecond();
        F.GNSS.nanosecond    = myGNSS.getNanosecond();
        F.GNSS.PosACC        = myGNSS.getPositionAccuracy(); //? not working :(
    }

    vTaskDelete(NULL);
}
```

### Expected behavior
To have a correct value returned by the _getPositionAccuracy()_ function. 

### Actual behavior
_getPositionAccuracy()_ function return a very strange result (like 107341414, even if my PDOP is quite near 1). 

`F.GNSS.PosACC        = myGNSS.getPositionAccuracy();`

When this line of code is commented I have noticed that everything run faster : 

Here it's when getPositionAccuracy() is used : 
[image]

And here it's when getPositionAccuracy() is not being used : 
[image]

(the delta time between each line is around 18ms for both example)

So does a maxdelay is being triggered and then the operation to get the position accuracy is cancelled ? 

Thank you, if the problem is only related to the fact that I'm using the NEO-M8N I'm really sorry for creating this issue...

Niv1900, 2021-01-19:
I'm really sorry to have polluted this git with a real false issue.

While reading [_SparkFun_u-blox_GNSS_Arduino_Library.h_](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/6d115b117b0fcc58249108685442e3cd5c7faf23/src/SparkFun_u-blox_GNSS_Arduino_Library.h#L964) this function is only valid for NEO-M8P and ZED-F9P. 

I need to use _getHorizontalAccuracy()_ and _getVerticalAccuracy()_.

closed 2021-01-19 by Niv1900

PaulZC, 2021-01-19:
Hi @Niv1900 ,

(We have a saying: "There is no such thing as a silly question." Please don't worry about asking questions. We are all learning!)

Your main issue is that getPositionAccuracy comes from the NAV-HPPOSECEF message (which is only supported on u-blox "high precision" modules).
Likewise getHorizontalAccuracy and getVerticalAccuracy come from NAV-HPPOSLLH (which is also "high precision").

getHorizontalAccEst and getVerticalAccEst come from the standard NAV-PVT message - which your NEO does support.

The reason you were seeing an extra delay in your code, is that NAV-PVT was being requested, then NAV-HPPOSECEF was being requested. The NAV-HPPOSECEF request would be NACK'd (Not ACKnowledged), which would take the extra time.

Please have a look at the [NAV-PVT Callback example](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Callbacks/CallbackExample1_NAV_PVT/CallbackExample1_NAV_PVT.ino). You may be able to use the UBX_NAV_PVT_data_t struct directly in your ESP code?

Good luck with your project,
Paul

Niv1900, 2021-01-19:
Thank you very much for your answer, everything is crystal clear now.

Oh indeed, the callback system looks nice, I have to play with it ! 

Looking forward to using this library, thank you @PaulZC.