sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #7: How to acess the flags in the data packet structures.

bvernham, 2021-01-21:
### Subject of the issue
Describe your issue here. If you reference a datasheet please specify which one and in which section (ie, the protocol manual, section 5.1.2). Additionally, screenshots are easy to paste into github.
You have a union of a struct and a variable inside another structure.  This is beyond my knowledge of C.

For example:  myGNSS.packetUBXHNRPVT->data.flags.gpsFixOK  How do I access gpsFixOK?  I have tried 10 different ways....
Thanks
### Your workbench
* What development board or microcontroller are you using?
* What version of hardware or breakout board are you using? 
* How is the breakout board wired to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

PaulZC, 2021-01-21:
Hi Bruce (@bvernham ),

You can access the individual bits of flags like this:
myGNSS.packetUBXHNRPVT->data.flags.bits.gpsFixOK

If you want to read the whole of flags, as a uint8_t, use:
myGNSS.packetUBXHNRPVT->data.flags.all

Best wishes,
Paul

closed 2021-01-21 by PaulZC

bvernham, 2021-01-21:
Yah, after staring at it for hours I figured it out.

Thanks

The union threw me but now I get it.

Also is there a particular reason why I don't want to use the call back?

I know that you have to use for example myGNSS.setAutoESFALGcallback to set up the call back but why don't I want to use it for the ESF?

The only reason why I would use ESF is prior to entering fusion mode but I think it is more efficient to use the call back and turn off the auto (which is why I asked the question when dead reckoning is entered.

Please let me know your thoughts.

Thanks

Bruce

Thanks