sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #11: Don't use delayMicroseconds()

MedadRufus, 2021-03-01:
### Subject of the issue
On the ESP32, the `delayMicroseconds()` function is a busy-wait. This is a problem with using Free-RTOS on the ESP32, as it holds up the processor, preventing it from doing anything else. On the other hand, the `delay()` implementation is non-blocking. This works because the RTOS tick frequency is once every millisecond(1 kHz).

In this library, there are 2 instances of `delayMicroseconds()`, both for 500 microseconds. I recommend making it 1 millisecond delay with the `delay()` function instead. Both are delays to wait before checking if bytes are available from u-blox GPS. I don't think there will be any issue waiting for a little bit longer before checking again.

The two locations of these delays are here:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/b1eb847f05a4fec11d0c30f1b091b44221d0aa41/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L2569
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/b1eb847f05a4fec11d0c30f1b091b44221d0aa41/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L2679

### Your workbench
* What development board or microcontroller are you using?
Custom dev board integrating ESP32 and U-blox MAX M8Q, connected over UART bus.
* What version of hardware or breakout board are you using? 
ESP32 + U-blox MAX M8Q
* How is the breakout board wired to your microcontroller?
UART connection
* How is everything being powered?
3.3V VCC powered via USB and regulated.
* Are there any additional details that may help us help you?
I am using FreeRTOS with the ESP32. I use the `CallbackExample1_NAV_PVT.ino` implementation in the Examples folder. [Link](examples/Callbacks/CallbackExample1_NAV_PVT/CallbackExample1_NAV_PVT.ino)

### Steps to reproduce
Create an RTOS thread to run the code on `CallbackExample1_NAV_PVT.ino`

### Expected behavior
It should be able to poll the GPS, but return back to doing other tasks while waiting for a response from GPS.

### Actual behavior
It is spending a long time waiting for the GPS to respond, holding up the processor from doing other things.

PaulZC, 2021-06-30:
Hi Medad (@MedadRufus ),
Thank you for your patience.
I've changed those 500us delays to 1ms as requested. It ties in nicely with the new SPI functionality. The new version of the library should go live in a day or two.
Code changes are [here](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/79f44129d0ea309eb1b54beb053e6d3e94c780d0).
Best wishes,
Paul

MedadRufus, 2021-06-30:
Excellent Paul. I have been running it with 1 ms delay for some time on the ESP32 and there are no issues.
Not tried with SPI though.

Shall I close the issue?

PaulZC, 2021-06-30:
Hi Medad,
I’ll close this when 2.0.8 goes live. Best leave it open until then - just in case.
All the best,
Paul

PaulZC, 2021-07-01:
Resolved in v2.0.8. Closing...

closed 2021-07-01 by PaulZC