sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #13: Performance with Apollo3 Core v2.x?

adamgarbo, 2021-03-24:
Hi @PaulZC,

Have you done any testing with v2.x of the Apollo3 Core and the u-blox library? I ran into a strange performance issue earlier today.

I first noticed it with the u-blox ZED-F9P and MicroMod Artemis Processor but to test, I switched to the u-blox SAM-M8Q and ran the "CallbackExample1_NAV_PVT". I ran the test for 10 minutes at a navigation frequency of 2 Hz. After a time, you can observe the processing of the data seems to bogs right down and jump from 0.5 s to a 6-second delay between messages. I created plots of the u-blox timestamps and those recorded by the Serial Monitor.

I didn't have debugging enabled at the time but am attempting to replicate the issue and will keep you posted.

Cheers,
Adam

**u-blox module timestamps:**
[image]

**Serial Monitor timestamps:**
[image]

PaulZC, 2021-03-27:
Hi Adam (@adamgarbo),
No, I still haven't been brave enough to try v2.0...
It's strange that it jumps to exactly six seconds? Have you made any changes to the example? Could you have other messages enabled (maybe configured in BBR)? I'm wondering if you are trying to generate more data than the port can handle?
Are you sure this is only present in v2.0 and not present in v1.2.1 too?
Cheers,
Paul

adamgarbo, 2021-03-30:
Hi @PaulZC,

I've decided to put aside working with Apollo3 Core v2.x until it's more stable, but to answer your questions, no changes were made to the example and the behaviour was only observed in v2.x. The u-blox ZED-F9P had recently been factory reset to defaults and configure just to output a minimum number of messages. The exact same behaviour was also observed with a u-blox SAM-M8Q, which seems to point to how the Artemis was handling the processing of the data. I did enable debugging the other day but can't remember exactly what message was presented. I believe it may have been a long repeating series of I2C timeouts.

I'm happy to close this issue for now and revisit it later on if it still persists.

Cheers,
Adam

closed 2021-03-30 by adamgarbo

PaulZC, 2021-03-30:
Thanks Adam,
Paul