sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #14: Request: setMeasurementRate(uint16_t msMeasurementPeriod)

nseidle, 2021-03-29:
Basically I'd like to be able to set the navFreq to something less than 1, such as 0.5 or 0.033 (for 30s between). I think the best way to do that is to add a set/getMeasurementRate() that directly sets the measurement rate.

See this [feature request](https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/7#issuecomment-802237654)

> What I am trying to do is slow the raw sample rate down to something less than 1 hz. For a lot of static surveying, there is no benefit from high-frequency sample rates ... 15 second, or even 30 second, sampling is fine, but we require longer observation periods (many hours to days). Thus, a helpful feature could include:
(1) setting slower sample rates (at the least, hard-wired options of 5, 15, and 30 seconds, but user specified sample period is more flexible)

PaulZC, 2021-03-30:
There are two ways to do this:
- add ```setMeasurementRate``` as you suggest - very happy to do that
- stick with (e.g.) 1Hz navigation rate but then change the message rate to a much slower rate

IIRC, I've tried this before and ran into problems when the interval was longer than around 45 seconds. I'm pretty sure that was using the first way. I'll go hunting for the code (it is probably in OpenLog Artemis).

The second way can be done two ways:
- If you want to keep using [configureMessage](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L3461) via [enableMessage](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L3486), then all you need to do is set the ```rate``` parameter to (e.g.) 30 to receive a message every 30 seconds (if the navigation rate is 1Hz)
- Because you are using the ZED, you can set the rate using a ```setVal8``` and the key [UBLOX_CFG_MSGOUT_UBX_RXM_RAWX_UART1](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/src/u-blox_config_keys.h#L535)

I'll report back once I find the other code. Watch this space!

PaulZC, 2021-03-30:
Again, because you are using the ZED, You can set the navigation rate manually if you want to using ```setVal16``` and the keys [UBLOX_CFG_RATE_MEAS](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/src/u-blox_config_keys.h#L651) and/or [UBLOX_CFG_RATE_NAV](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/src/u-blox_config_keys.h#L652)

closed 2021-03-30 by PaulZC

nseidle, 2021-04-12:
This is working great! Thank you.