sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #19: Examples display incorrect fractional lat/long

nseidle, 2021-03-31:
If the lat_frac has leading zeros, they are dropped, leading to the incorrect value being printed:

[image]

I know examples 10 and 11 are effected but there may be others:

[image]

[image]

One solution is to use printf

  Serial.printf("%d.%07d\n", lat_int, lat_frac); // Print the integer part of the latitude

But I know its support is iffy. We might have to break down and just write a silly helper function inside these examples.

PaulZC, 2021-04-01:
Fixed in [this commit](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/e483b119cc254833419d528c9b23c520475d2dbc).
I've searched, and I think this is one of a kind...
A couple of examples weren't using nanoseconds correctly (nanoseconds can be negative) so I've corrected those too, but that's a different issue.

PaulZC, 2021-04-01:
I think this doesn't actually affect Example11 as the printing there is done all in one go?

closed 2021-04-01 by PaulZC