sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #20: Fix fileBuffer RAM allocation

adamgarbo, 2021-03-31:
Hi @PaulZC,

As discussed earlier, we'll need to fix howthe  `.createFileBuffer()` function allocates RAM. At present, if the system makes multiple calls to `.begin()`, RAM usage will eventually exceed the available memory (e.g. a 16 KB file buffer will consume all 384 KB RAM after 24 sleep/wake cycles). In actuality, this seems to occur at about the 14th sleep/wake cycle, which causes an immediate hard reset of the processor.

Offending line of code:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/7f38fcdae96691c1515b54cd541b53844ca4201e/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L3084

Cheers,
Adam

PaulZC, 2021-04-01:
Thanks for finding this Adam!
Corrected in [this commit](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/2ce23fe6960101fdfd8716510de2c5e2970addb2).
Please give Example26 a try.

adamgarbo, 2021-04-01:
Thanks, Paul! 

Will try it out this morning. Can you clarify what you mean by "End will stop all automatic message processing and free (nearly) all used RAM."? Is there still some RAM that will be lost after each `.begin()` call? Thinking about long-term (1-year) deployments of the code.

Cheers,
Adam

PaulZC, 2021-04-01:
Hi Adam,
The memory leak - from multiple calls to .begin - should be completely fixed.
.end is optional but you can use it to deallocate (delete) all of the allocated RAM _except_ payloadCfg. (That's why I use the word "nearly"!).
Cheers,
Paul

adamgarbo, 2021-04-01:
Thanks, Paul!

I've been letting the code run all morning and am currently up to 40 sleep/wake cycles with a 16 KB buffer and no issues!

Cheers,
Adam

PaulZC, 2021-04-01:
Jolly good!
Have a nice Easter!
:-)

closed 2021-04-01 by PaulZC