sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #22: Power Save Mode on NEO-M9N

JMyrng, 2021-04-01:
### Can´t find a way to enter PSMOO (Power Save Mode On/Off)
Is there a way to enter the mode with this library? The mode is described in the [NEO-M9N integration manual](https://www.u-blox.com/sites/default/files/NEO-M9N_Integrationmanual_%28UBX-19014286%29.pdf) in section 3.9.2. 

### Your workbench
* What development board or microcontroller are you using?
Arduino MKR1310
* How is the breakout board wired to your microcontroller?
I2C with jumper wires 
* How is everything being powered?
LiPo Battery

### Things I have tried
I tried example 18, but due to the protocol version it did not work. 

Is `setVal` the way to go? 

Thanks in advance for your help.

PaulZC, 2021-04-01:
Hi @JMyrng ,
Yes, have a look at section 5.9.16 in
https://www.u-blox.com/en/docs/UBX-19035940
You need setVal8 and CFG_PM_OPERATEMODE. It looks like we don’t yet have that key included in config_keys.h so you may need to use the full hex value. The settings are listed at the end of that section. Set to 1 for PSMOO.
Let me know if you need any more help.
Cheers,
Paul

JMyrng, 2021-04-02:
Hi @PaulZC 
Thank you very much for your advice. I have tried it like this:
```bool setValueSuccess;
  setValueSuccess = myGNSS.setVal8(0x20d00001, 1);
  if (setValueSuccess == true)
    Serial.println("OperateMode was successfully set to PSMOO (Power Save Mode On/Off)");
  else
    Serial.println("OperateMode set failed"); 
```

And it worked as desired. Without further configuration, the power consumption is already significantly lower. 

[image]

closed 2021-04-02 by JMyrng