sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #24: getNavigationRate() fails after setNavigationRate()

nseidle, 2021-04-12:
This is not really a bug, so much as a report of weird behavior of the ZED. If you do a getNavigationRate() directly after a setNavigationRate() it will report the wrong value. It's not the ZED storing value or delaying; adding a 5s delay doesn't fix the problem, there's something else going on. Reading a second time (even immediately) always reports the correct value. 

I've seen this on RTK Surveyor when setting the meas/nav rates to larger values (30000 / 41 for 1234s between reports). The work around is always doing a 2nd getNavigationRate() before trusting it.

    #include <Wire.h> //Needed for I2C to GNSS
    
    #include <SparkFun_u-blox_GNSS_Arduino_Library.h> //http://librarymanager/All#SparkFun_u-blox_GNSS
    SFE_UBLOX_GNSS myGNSS;
    
    void setup()
    {
      Serial.begin(115200);
      while (!Serial); //Wait for user to open terminal
      delay(250);
      Serial.println("SparkFun u-blox Example");
    
      Wire.begin();
    
      if (myGNSS.begin() == false) //Connect to the u-blox module using Wire port
      {
        Serial.println(F("u-blox GNSS not detected at default I2C address. Please check wiring. Freezing."));
        while (1);
      }
    
      myGNSS.setI2COutput(COM_TYPE_UBX); //Set the I2C port to output UBX only (turn off NMEA noise)
    
      uint16_t currentNavigationRate = 0;
    
      myGNSS.setNavigationRate(8);
    
      //delay(5000); //You can delay as long as you'd like, it doesn't fix the problem
    
      currentNavigationRate = myGNSS.getNavigationRate();
      Serial.printf("currentNavigationRate (should be 8): %d\n", currentNavigationRate);
    
      //The ZED-F9P will report an incorrect nav rate if we have rececently changed it.
      //Reading a second time insures a correct read.
      currentNavigationRate = myGNSS.getNavigationRate();
      Serial.printf("Second read (should be 8): %d\n", currentNavigationRate);
    
      myGNSS.setNavigationRate(4);
    
      currentNavigationRate = myGNSS.getNavigationRate();
      Serial.printf("currentNavigationRate (should be 4): %d\n", currentNavigationRate);
    
      currentNavigationRate = myGNSS.getNavigationRate();
      Serial.printf("Second read (should be 4): %d\n", currentNavigationRate);
    }
    
    void loop()
    {
    
    }

PaulZC, 2021-04-12:
Oh... Thanks! I've seen this too!

I saw it when testing the new [Measurement and Navigation Rate Example](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Example25_MeasurementAndNavigationRate/Example25_MeasurementAndNavigationRate.ino).

I saw the exact same thing, that [getMeasurementRate](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Example25_MeasurementAndNavigationRate/Example25_MeasurementAndNavigationRate.ino#L89) would return the old value, but the [getNavigationRate](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Example25_MeasurementAndNavigationRate/Example25_MeasurementAndNavigationRate.ino#L93) would return the new updated value.

I wasn't sure if the module was actually returning the old value, or whether the old value was being buffered by [getNavigationFrequencyInternal](getNavigationFrequencyInternal). My fix was to make sure that the UBX CFG RATE data was [marked as stale](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Example25_MeasurementAndNavigationRate/Example25_MeasurementAndNavigationRate.ino#L85), forcing the library to request fresh data when calling getMeasurementRate. Calling either getMeasurementRate or getNavigationRate twice will do the same thing, ensuring fresh data is requested from the module by the second call.

PaulZC, 2021-04-12:
@nseidle : Also, just in case you missed it, I added a new feature which means it is now possible to set different "auto" message rates for individual messages. E.g. You can set different rates for the PVT and RAWX / SFRBX messages if you want to. Have a look at [Example27](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Example27_MultipleRates/Example27_MultipleRates.ino) for more info. Enjoy!

PaulZC, 2021-04-12:
OK. I now understand the root cause of this issue...

When you call ```setMeasurementRate``` or ```setNavigationRate```, the library [polls the module for the CFG RATE](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L9144-L9151), the navigation rate or measurement rate setting is changed and the data is written back to the module. **BUT!** The data that the library polled from the module will still be stored inside the library and will still be marked as fresh. So when you call ```getMeasurementRate```, the old / stale value is returned. I need to make sure that the measurement rate data is marked as stale inside ```setMeasurementRate``` _and_ ```setNavigationRate``` so that ```getMeasurementRate``` _or_ ```getNavigationRate``` is forced to request fresh data!

I'll fix this in the morning...

But I also need to double-check that this is the only place where this happens / can happen. And that is definitely a job for the morning...!

nseidle, 2021-04-12:
Excellent. Once you posted your first response, I figured it was likely the lib and not the module. Agreed, it looks like stale data rather than ZED issue.

PaulZC, 2021-04-13:
Resolved!
Please pull the [release_candidate branch](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/release_candidate) for the [latest changes](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/63d8b60f4271da1184179228a1680e474037c28f). I'll merge and release as soon as you are happy with the NMEA-logging.

nseidle, 2021-04-13:
Works for me! Thanks!

closed 2021-04-13 by nseidle