sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #27: How to get fullyResolved status?

ZaneL, 2021-05-03:
Hello,

I want to emulate the behavior of this library:
[https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/libraries/GNSS](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/libraries/GNSS)

Specifically, the code that looks like this:

```
if ((myLocation.fixType() == GNSSLocation::TYPE_3D) && (myLocation.ehpe() <= 30.0) && myLocation.fullyResolved())
{
 // suspend GPS 
}
```

Using your library, I guess this would look something like the following?

```
if ( (myGNSS.getFixType() == 3) && (getHighResLongitudeHp() <= 30) && fullyResolvedFlagCheck )
{
  myGNSS.powerOff(0);
}
```

I can see the flag here, but I'm not really sure if there is a good way to query it...

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/2c71ab719eebd22f9098a2568839d20b8429f192/src/u-blox_structs.h#L360

PaulZC, 2021-05-04:
Hi @ZaneL ,

(Nice to meet you! Thanks for the work you did on your Teensy ICM-20948 library and visualization tool. We like that a lot.)

If you need access to a particular data field, and it does not have a helper function, then you can simply read the value from the appropriate 'packet' in memory. In your case, fullyResolved can be read using:

```
uint8_t fullyResolved = myGNSS.packetUBXNAVPVT->data.valid.bits.fullyResolved;
```

You will need to make sure that the data is fresh, not stale. In the helper functions, we do that using the ```moduleQueried``` flags. ```getHorizontalAccEst``` will do that for you. So long as you call ```getHorizontalAccEst``` (or ```fixType```) before you read fullyResolved, you can be confident that the data is fresh.

Best wishes,
Paul

ZaneL, 2021-05-04:
Thanks Paul! ...and thanks for writing this library - looks like a lot of work!

I ended up using your other functions as a reference and just added this simple function:
```

//Check to see if the UTC time has been fully resolved
bool SFE_UBLOX_GNSS::getTimeFullyResolved(uint16_t maxWait)
{
  if (packetUBXNAVPVT == NULL) initPacketUBXNAVPVT(); //Check that RAM has been allocated for the PVT data
  if (packetUBXNAVPVT == NULL) //Bail if the RAM allocation failed
    return (false);

  if (packetUBXNAVPVT->moduleQueried.moduleQueried1.bits.fullyResolved == false)
    getPVT(maxWait);
  packetUBXNAVPVT->moduleQueried.moduleQueried1.bits.fullyResolved = false; //Since we are about to give this to user, mark this data as stale
  packetUBXNAVPVT->moduleQueried.moduleQueried1.bits.all = false;
  return ((bool)packetUBXNAVPVT->data.valid.bits.fullyResolved);
}
```

I think something like that might be useful enough to add into the official library? 

Also I have one other question. Essentially what I'm trying to do is get a good fix, power off the GNSS module and then wake it back up in 30 minutes by toggling INT0. This is all working perfectly, but I guess I have a question about this "hot start" functionality. In the datasheet it seems to specify that if you pass the GNSS module lots of information when it wakes back up, this decreases the duration of this hot start fix. 

I can see in the code that this is how it powers off the module:
```
  // Power off device using UBX-RXM-PMREQ
  packetCfg.cls = UBX_CLASS_RXM; // 0x02
  packetCfg.id = UBX_RXM_PMREQ;  // 0x41
  packetCfg.len = 8;
  packetCfg.startingSpot = 0;
```

...which I presume is saving the ephemeris data in the module's RAM, so as long as the V_BACKUP pin is powered, when the module comes out of sleep mode it can do a hot start. Is it necessary to do anything else, or is that the extent of this hot start functionality? For example after I wake it back up to I need to send the module the data I pulled out of it before it went to sleep?

PaulZC, 2021-05-05:
Hi @ZaneL ,

Yeah... A **lot** of work has gone into this library over the years. Nate started it all and we've had many contributions from the community... Speaking of which, please send us a Pull Request and we'll be happy to include your fullyResolved helper function. Contribution details are [here](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/CONTRIBUTING.md). Please do add example code for it too. You don't need to add a whole new example, just add your call to one of the existing examples, possibly Example24? Please also add your new function to keywords.txt.

Hot start is something we need to investigate. We are working on a new design which needs to get a fix as quickly as possible. I'll open a separate issue for that.

Best wishes,
Paul

closed 2021-05-05 by PaulZC