sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #29: Weak definition of processNMEA/processNTRIP

Gatchers, 2021-05-04:
### Subject of the issue
Weak definitions of processNMEA, processNTRIP

### Your workbench
Experimenting with different compilers

### Expected behavior
I can override processNMEA, processNTRIP with my own version, the linker favouring my non-weak function.

### Actual behavior
As the header defines the function as weak, the compiler sees the function code as weak, leaving the linker to decide what function to use. With an ARM Compiler set up, typically my replacement function is ignored, and the library version is used. Would it be better to prototype the function as normal (without the weak attribute) and then define the library function like so?

__attribute__((weak)) void SFE_UBLOX_GNSS::processNMEA(char incoming)

PaulZC, 2021-05-04:
Hi Gareth (@Gatchers ),
In principle I don't have a problem with changing the way we define ```processNMEA``` as weak, **so long as we can maintain backward compatibility**. The ```__attribute__((weak))``` in the .h has been there for a lot of years and I think you're the first person to suggest this, but I'm not against your suggestion. If you can demonstrate that Example2_NMEAParsing still compiles and runs on all the popular Arduino boards, then I will certainly consider merging a Pull Request from you.
Best wishes,
Paul

PaulZC, 2021-05-19:
Closing... Please reopen if you want to take the discussion further.
Best wishes,
Paul

closed 2021-05-19 by PaulZC