sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #31: [Feature Request] Hot-Start Support

PaulZC, 2021-05-05:
As mentioned in issue #27, it would be useful to have an example and/or code support for hot start (when waking from sleep)

UT2UH, 2021-06-21:
Any updates on the right approach to hot-start speeding up?

PaulZC, 2021-06-27:
Hi @UT2UH ,
Sorry - no - not yet... Still on the back-burner for now. But we will get to this as a new product we are working on will need it.
Best wishes,
Paul

PaulZC, 2021-10-17:
Hi @JMyrng ,

The first board will be based on ZOE-M8Q but, yes, an M10 will follow soon after. Low power operation has been designed in from the start, and the sleep current is very low. We will publish the exact figure when the new boards are launched, but I think you will be pleased!

It is probably not clear / obvious, but ```setAutoPVT``` will not work with ```powerOff```... ```setAutoPVT``` instructs the module to automatically generate PVT messages, without polling. Once autoPVT is enabled, ```getLatitude()``` does not poll the PVT data and so there is no message generated which would cause the module to wake up... Please try calling (e.g.) ```getNavigationRate()```. That will generate a poll request causing the module to wake.

Best wishes,
Paul

JMyrng, 2021-10-17:
I'm really looking forward to the M10 module! Please consider breaking out the Vbackup and the enable pins. 

Thanks for that clarification! Maybe that should be noted in the examples/ code? I can't test it at the moment because I've changed my code too much, but I'm sure it will work if I need it again (e.g. on the M10). 

Best regards,
JMyrng

PaulZC, 2021-12-01:
Hi @UT2UH / @JMyrng ,

Please see the [release notes for v2.1.0](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/releases/tag/v2.1.0) - we've added full support for AssistNow Online, Offline and Autonomous to deliver very quick Time-To-First-Fixes. Please give the new examples a try - I think you will be very pleased. Please raise a new issue if you see any gremlins!

Enjoy!

Best wishes,
Paul

closed 2021-12-01 by PaulZC