sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #34: Getting vertical velocity (vVel) value

maxmclau, 2021-05-05:
Super impressed with this library! Currently I'm computing vertical velocity of my ZOE-M8Q over a series of altitude readings and was curious if the uBlox modules provide vertical velocity as part of their PVT solutions the same way they provide directional velocities? Looking through their Interface Manual I was only able to find reference to it in their proprietary PUBX messages (**Section 31.3.2**).

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/2c71ab719eebd22f9098a2568839d20b8429f192/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L10010

I see that this library exposes the NED `velD` values (**Section 32.14.1**), though as I understand it that is computed through IMU data in modules like the ZED-F9K.

Sorry, this isn't an issue with the library necessarily - let me know if I should bring this question over the uBlox forums if it's off topic.

PaulZC, 2021-05-06:
Hi Maxmillion (@maxmclau),

The best place to ask questions like this is in the [GPS Forum](https://forum.sparkfun.com/viewforum.php?f=116), but we forgive you... ;-)

```getNedDownVel``` extracts the down velocity from the basic UBX NAV PVT message, which your ZOE will support. Please give it a try - it should give you exactly what you need. The velocity is mm/s and is int32_t (signed).

There is also a dedicated UBX NAV VELNED message which the library supports too. But there are no helper functions for that message as yet, so you will need to extract the velocity manually. The following should work just fine:

```
int32_t velD = 0;
if (myGNSS.getNAVVELNED()) // Returns true if fresh NAV VELNED data is received
{
  velD = myGNSS.packetUBXNAVVELNED->data.velD; // velD is cm/s in NAV VELNED (not mm/s)
}
```

Check out the library examples for more details.
Best wishes,
Paul

closed 2021-05-06 by PaulZC

maxmclau, 2021-05-06:
Appreciate the help help Paul!