sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #37: GPS-RTK2 with correction source has up to 2m deviation in latitude

geeks-r-us, 2021-05-09:
I connected the GPS-RTK2 board to an u-blox high precision antenna hooked up an HC-06 Bluetooth module to forward correction data via android app to the module and attached an esp32 microcontroller with LoRa to transmit the position. 

To gather the information form the module we just took the example codes from this repository.

Now we used a commercial tachimeter from Leica, which should deliver values with a precision of a few mm,  to determine the exact position of our antenna.

When we compared the values from Leica with them of u-blox longitude matches perfect but latitude has a deviation up to 2m.
Reported accuracy from the module was in the range for 6-30cm but not meters. 

What I would expect is a deviation in both longitude and latitude but only in one is a bit strange?

Are there known issues that could cause a behavior like that?

PaulZC, 2021-05-09:
Hi @geeks-r-us ,
Can you give us a bit more information about how you collected the data with the GPS-RTK2? Did you use one of the survey-in (RTCM Base) examples, and how long did you define for the survey-in period? There is a tutorial [here](https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station) which should help - although it uses a different method to establish the antenna position. [This tutorial](https://learn.sparkfun.com/tutorials/setting-up-a-rover-base-rtk-system)  does discuss survey-in.
But, I have to say, this does not sound like an issue with the library itself. It seems more likely to be linked to the subtleties of how you collected your data. It would be worth asking your question in the [GPS Forum](https://forum.sparkfun.com/viewforum.php?f=116) too.
Best wishes,
Paul

PaulZC, 2021-05-12:
Closing... Please reopen if you need more suggestions.

closed 2021-05-12 by PaulZC