sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #38: nmea parsing not working smooth

mstranne, 2021-05-21:
Hey,

I got an m8p connected to an ESP32 via I2C. 
In the old library,  parsing nmea messages in combination with the micronmea lib worked fine.

However, at the moment I do not get new (valid) nmea sentences frequently.

I got setNavigationFrequency to 10, disabled all messages except GGA, setI2CTransactionSize to 128 and call checkUblox frequently.

Then I process new nmea sentences like this:

```
void SFE_UBLOX_GNSS::processNMEA(char incoming)
{
  if(nmea.process(incoming) == true) {  //new full sentence
    const char* sentence = nmea.getSentence();
    if(nmea.isValid() == true){
      //do something
    }   
  }
}
```
 
I am also using pushRawData  to feed  rtcm corrections received  to the the ublox chip .

I  expect new new sentences at 10Hz.  Any idea  where the  problem could be ?

Cheers Marco.

mstranne, 2021-05-21:
when I do not use the rtcm stream and pushRawData it works better, however sometimes there are also 3 to 5 sec without an new nmea update.

PaulZC, 2021-05-21:
Hi Marco (@mstranne ),

Did 10Hz work OK with the old library? (I'm trying to work out what has changed.)

Looking at the datasheet for the NEO-M8P, it suggests you can only achieve 10Hz (PVT) or 8Hz (RTK) when tracking GPS only. Have you disabled the other constellations?

[image]

```enableGNSS``` will help here:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/b20579e279000b6ff6d959341b25be04b5b24d46/examples/Data_Logging/DataLoggingExample5_Fast_RXM/DataLoggingExample5_Fast_RXM.ino#L147-L156

Best wishes,
Paul

mstranne, 2021-05-21:
Damn, I thought about something like that.

However, the problem was that I enabled high precision mode and as soon as the correction data was forwarded to the ublox chip, and it got an RTK fix the nmea messages have a length of 88, which was more than maxNMEAByteCount (82), so messages were not parsed any more.

Example String:
$GPGGA,172814.0,3723.46587704,N,12202.26957864,W,2,6,1.2,18.893,M,-25.669,M,2.0,0031*4F

Would be awesome if you could fix that value!

Thanks a lot.

Best wishes,
Marco

Darmann2, 2021-06-27:
Hello!

I run up to the same problem, and I think this is a little bug not a feature request.

GGA without RTK is working:
$GNGGA,123303.50,4648.8589739,N,01451.3921055,E,1,12,0.63,468.064,M,42.859,M,,*4C

GGA with RTK is not working
$GNGGA,080102.75,4648.8577377,N,01451.3912508,E,5,12,0.56,464.699,M,42.859,M,0.8,0000*6E

Thanks!
Alfred

PaulZC, 2021-06-27:
Hi Alfred (@Darmann2 ),

Nah, I'm calling it a feature request ;-D . The extra precision from RTK takes the length of the GGA message beyond the usual maximum length for NMEA (82 bytes). I'm trying to decide what new maximum length to use (probably 128 bytes just to make sure we don't run into this issue again) or whether to add a new function to let users define what maximum length to use. If you have a preference, please let me know.

A fix is coming soon. Meanwhile, if you want to fix this yourself, please edit this line in the library header file:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/929a9ee6a27a34b653a1c42031cba66d395b1598/src/SparkFun_u-blox_GNSS_Arduino_Library.h#L1322

Best wishes,
Paul

PaulZC, 2021-06-29:
Hi Marco (@mstranne ) / Alfred,
Thank you for your patience. I've changed ```maxNMEAByteCount``` to 88 and have made it configurable via ```setMaxNMEAByteCount```. The code changes are [here](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/28c71ae49c826f406d38c0f1dbee1df01af762ba). Version 2.0.8 of the library will be released soon once I have finished testing the new SPI functionality.
Very best wishes,
Paul

PaulZC, 2021-07-01:
Resolved in v2.0.8. Closing...

closed 2021-07-01 by PaulZC