sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #40: GPS non-responsive, checkUbloxI2C: OK, zero bytes available

abishur, 2021-06-17:
### Subject of the issue
Using the autoPVTcallback and the AutoNAVODOcallback my program as recently begun to get no response back.  Enabling debugging mode shows that it checks but returns with the message "checkUbloxI2C: OK, zero bytes available".  Currently I execute a checkUBlox() followed by a checkCallbacks() once a second,I have tested to see if checking for a call back each scan cycle and only checking for callbacks once a second will resolve my issue, but the issue remained.  Time for freezing to occur is random.  Sometimes it takes a few minutes, sometimes about an hour and a half.  The rest of the program will run as normal, the only part that freezes is the GPS communication.  Other devices on the I2C bus continue to communicate

### Your workbench
* I am using an ESP32
* The ZOE-M8Q is connected to my ESP32 via I2C and is configured to only send UBX messages on I2C, it shares the bus with other devices.  I have tried to remove the other devices but issue remains.  All devices are operating at 3.3v.  Prior to the issues beginning I did have the serial port wired to a raspberry pi sending out NEMA sentences, but I disconnected that... shortly before the issues began. 
* GPS board is being powered from 3.3v provided by ESP32, ESP32 is powered by USB connection used for programming

### Steps to reproduce
Code I use to initialize GPS
// Try to connect to GPS device unless it's already been successfully initiated
```
  if (myGPS.begin(Wire) != false)
  {
    Serial.println(F("GPS Connected"));
    GPScon = true;
    myGPS.enableDebugging();
    myGPS.setUART1Output(COM_TYPE_NMEA); // Set the Serial Port to output NMEA only for Pi Time source
    myGPS.setI2COutput(COM_TYPE_UBX); //Set the I2C port to output UBX only (turn off NMEA noise)
    myGPS.setNavigationFrequency(5); //Produce 5 solutions per second
    myGPS.setAutoPVT(true); // Tell the GPS to "send" each PVT solution automatically
    myGPS.setAutoNAVODO(true); // Tell the GPS to "send" each Odometer solution automatically

    uint8_t customPayload[MAX_PAYLOAD_SIZE];

    // By default the Odometer is disabled, in case configuration is lost, when device is turned on, enabled odometer.
    ubxPacket customCfg = {0, 0, 0, 0, 0, customPayload, 0, 0, SFE_UBLOX_PACKET_VALIDITY_NOT_DEFINED, SFE_UBLOX_PACKET_VALIDITY_NOT_DEFINED}; // Prepare ubx packet
    customCfg.cls = UBX_CLASS_CFG; // This is the message Class
    customCfg.id = UBX_CFG_ODO; // This is the message ID
    customCfg.len = 0; // Setting the length to zero let's us poll the current settings
    customCfg.startingSpot = 0; // Always set the startingSpot to zero (unless you really know what you are doing)
    uint16_t maxWait = 250;
    if (myGPS.sendCommand(&customCfg, maxWait) == SFE_UBLOX_STATUS_DATA_RECEIVED) // We are expecting data and an ACK
    {
      customPayload[4] = byte(5);  // Odometer and low-speed COG filter flags (1st bit = use odometer, 2nd bit = useCOG, 3rd = OutLPVel, 4th = outLPCog)
      customPayload[5] = byte(3); // Odometer profile (0 = Running, 1 = Cycling, 2 = Swimming, 3 = Car, 4 = Custom)
      customPayload[12] = byte(30); // Speed below which Course-over-ground (COG) is computer with low-seed COG filter (divided by 10 so 15 = 1.5 meters/second)
      customPayload[13] = byte(50); // Maximum acceptable position accuracy for computer COG with the low-speed COG filter
      customPayload[16] = byte(10); // Velocity Low-Pass filter level, range 0 (heavy filter) to 255 (weak filter)
      customPayload[17] = byte(10); // COG Low-Pass filter level at speeds < 8 m/s.  Range 0 (heavey filter) to 255 (weak filter)
      if (myGPS.sendCommand(&customCfg, maxWait) != SFE_UBLOX_STATUS_DATA_SENT) // We are expecting only an ACK
      {
        Serial.println(F("sendCommand (poll / get) failed!"));
      }

      myGPS.setAutoPVTcallback(&PVTupdate); // Enable automatic NAV PVT messages with callback to printPVTdata
      myGPS.setAutoNAVODOcallback(&ODOupdate); // Tell GPS to "send" Odometer readings automatically
      // Save the settings to battery backup
      myGPS.saveConfigSelective(VAL_CFG_SUBSEC_IOPORT);
    }
  } else {
    Serial.println(F("Error connecting to GPS"));
  }
```
Code I use in callback function:
```
void PVTupdate(UBX_NAV_PVT_data_t ubxDataStruct)
{
  // Start by checking if data is valid
  Serial.println("Callback!");
  gpsFix = bitRead(ubxDataStruct.flags.all, 0);

  //int psmState = ubxDataStruct.flags.all & 0x1C; // Power Save Mode is store in bits 2,3, and 4, and it with 0x1C (and probably shift bits 2 in some direciton) to get PSM state

  // Only Update when we have a valid GPS Fix
  if (gpsFix)
  {
    gpsFixType = ubxDataStruct.fixType;
    headingFix = bitRead(ubxDataStruct.flags.all, 5);
    if (gpsFixType == 3)
[... 158 lines cut ...]
packetAck now valid
packetAck classAndIDmatch
process: incoming "automatic" message: Class: 0x1 ID: 0x9
processUBX: incoming "automatic" message: Class: 0x1 ID: 0x9
Incoming: Size: 20 Received: CLS:NAV ID:0x9 Len: 0x14 Payload: 0 0 0 0 78 AB 40 19 0 0 0 0 4D 1 0 0 0 0 0 0
packetCfg now valid
packetAck now valid
packetAck classAndIDmatch
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: OK, zero bytes available

Sending: CLS:NAV ID:0x10 Len: 0x0 Payload:
sendCommand: Waiting for ACK response
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 10 bytes
Incoming: Size: 2 Received: CLS:ACK ID:0x1 Len: 0x2 Payload: 1 10
packetCfg now valid
packetAck now valid
packetCfg classAndIDmatch
packetAck classAndIDmatch
waitForACKResponse: no data and valid ACK after 70 msec
```

abishur, 2021-06-18:
I have done more testing and I am getting this same issue using the example files "Example1_AutoPVT" and "CallbackExample1_NAV_PVT".  However, "Example15_GetDateTime" has been running for a while without any issues.  The problem seems to specific to the auto option.

PaulZC, 2021-06-18:
Hi @abishur ,

I don't have time to go through your code line by line, but there are some things in there that could cause a problem.
- You say you call checkCallbacks() once a second but you have the navigation frequency set to 5Hz. I recommend calling both ```checkUblox()``` and ```checkCallbacks()``` every 50ms - like in CallbackExample1_NAV_PVT.
- It looks like you might have been experimenting with power save mode, but you have the navigation frequency set to 5Hz. That sounds like a bad combination.

Here is what I suggest:

**Factory Defaults**
Please reset your module back to the factory defaults, just in case there is anything in the battery-backed settings which is causing a problem. You can do this by calling ```myGNSS.factoryDefault(); delay(5000);``` after ```myGNSS.begin();```

**I2C Pull-ups**
We know that the u-blox modules work best on I2C when there are no other pull-up resistors on the bus, and when the bus is running at 100kHz (not 400kHz). The module has internal pull-ups which work perfectly. We routinely open all of the pull-up jumpers on our u-blox boards. Removing the other devices while you test the u-blox module on its own is a good idea. Maybe you can move the other devices to a second I2C bus for your project?

Please try CallbackExample1_NAV_PVT again - with the factory defaults. Do you still see the issue?

Best wishes,
Paul

PaulZC, 2021-06-27:
HI @abishur ,
How is it going? Are you still seeing this issue?
I know you are using I2C, not Serial, but I am wondering if this could be related to issue #42 ?
Best wishes,
Paul

PaulZC, 2021-06-30:
Closing... Please re-open @abishur if you are still having problems.
Best wishes,
Paul

closed 2021-06-30 by PaulZC