sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #42: Missing gps-samples because of wrong checksum

TheXtremeKing, 2021-06-21:
###Subject of the issue
Missing gps-samples because of wrong checksum.

Does anyone else have this issue?
If not then maybe it's just my unit that is broken / I've a very nasty bug in my code somewhere but it is so strange the resulting error is so specific that I could code it out...

### Your workbench
Espressif ESP32 Dev Module 
Platform IO / vs-code
gps: sam-m8q-0-10 wired via serial to the esp32

### Steps to reproduce
Running the library setup like this:
```
    UbloxGps.setDynamicModel(dynModel::DYN_MODEL_BIKE); 
    UbloxGps.setUART1Output(COM_TYPE_UBX);             
    UbloxGps.setNavigationFrequency(10);    
    UbloxGps.setAutoPVT(true);                          

    //enable/disable gps-systems
    UbloxGps.enableGNSS(true, SFE_UBLOX_GNSS_ID_GPS);     
    UbloxGps.enableGNSS(false, SFE_UBLOX_GNSS_ID_GALILEO);
    UbloxGps.enableGNSS(true, SFE_UBLOX_GNSS_ID_GLONASS); 
    UbloxGps.enableGNSS(false, SFE_UBLOX_GNSS_ID_SBAS);  
    UbloxGps.enableGNSS(false, SFE_UBLOX_GNSS_ID_BEIDOU); 
    UbloxGps.enableGNSS(false, SFE_UBLOX_GNSS_ID_IMES);  
    UbloxGps.enableGNSS(false, SFE_UBLOX_GNSS_ID_QZSS);
```   

Calling GetPVT() fast enough and when it returns true, store gps data and check for missing samples (we should get a sample every 100ms).

### Expected behavior
no missing samples

### Actual behavior
once in a while (around a minute) a sample gets discarded because of an checksum error.

###Work around
change the checksum check under certain conditions in SFE_UBLOX_GNSS::processUBX() like this:
```
  //orignal code
 else if (incomingUBX->counter == incomingUBX->len + 5) //ChecksumB
  {
    incomingUBX->checksumB = incoming;
    currentSentence = NONE; //We're done! Reset the sentence to being looking for a new start char

    //Checksum black magic (I mean workaround)
    if (incomingUBX->checksumA == 0 && (incomingUBX->checksumA != rollingChecksumA))
    {
      Serial.print(" fixing checksum, original: ");
      Serial.print(rollingChecksumA);
      Serial.print("/");
      Serial.println(rollingChecksumB);

      rollingChecksumB = rollingChecksumA+1;
      rollingChecksumA = 0;
    }
}
```

So it seems that if the checksum A-part is 0 then sometimes (not always) something gets broken...
I've also noticed that after this occurence an extra byte is received which is discarded in the first case statement of SFE_UBLOX_GNSS::process() 

This is a log with terminal info showing a fixed sample: 

_time: 18:38:08.100 
_speed: 116 _heading: 16977017 _lat: 53 _lon: 70 _alt: 9943
_fix: true _HDOP: 0 _VDOP: 0 _PDOP: 0 _SDOP: 547 _Sats: 14

 fixing checksum, original: 67/28  >>checksum calculated, this gets "fixed" by my workaround
Incoming: Size: 92 checksumA: 0 checksumB: 68 Received: CLS:NAV ID:PVT Len: 0x5C Payload: 98 51 26 9 E5 7 6 15 12 26 8 37 18 0 0 0 96 E5 F2 B 3 1 A E 39 D1 33 4 6F 7C B1 1F 95 D9 0 0 6F 27 0 0 C0 1E 0 0 4E 2C 0 0 70 0 0 0 FF F7 FF FF FF 1C 0 0 0 71 0 0 0 79 C 3 1 9E 1 0 0 68 C3 27 0 8E 0 0 0 E0 4A 23 0 0 0 0 0 0 0 0
packetCfg now valid
packetAck now valid
packetCfg classAndIDmatch
packetAck classAndIDmatch
nonsens received: 26   >>extra byte received that doesn't belong to any packet anymore
Incoming: Size: 18 checksumA: 203 checksumB: 186 Received: CLS:NAV ID:0x4 Len: 0x12 Payload: IGNORED
packetCfg now valid
packetAck now valid
packetAck classAndIDmatch
*-*
_time: 18:38:08.200 
_speed: 28928 _heading: 51149056 _lat: 53 _lon: 70 _alt: 10095
_fix: true _HDOP: 0 _VDOP: 0 _PDOP: 0 _SDOP: 105985 _Sats: 14

nseidle, 2021-06-21:
I'm going to wager some guesses here because @PaulZC is on a much deserved vacation this week.

It's tricky to troubleshoot your issue without logic traces. You can spend a lot of time head scratching but I recommend you hookup a logic analyzer ([they're](https://www.sparkfun.com/products/15033) cheap!) and verify what the SAM is putting out is what your ESP32 says it is receiving. My guess is that the SAM is behaving itself and your ESP32 is being interrupted while your RX buffer is filling up which is corrupting the incoming NMEA/UBX. 

I have hit slightly different issues with the ESP32 that may help. One solution I came across created a task on core 0 so that Serial.begin() started on core 0 (which leads all the UART interrupts to occur on core 0). You can see that [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/Tasks.ino#L108).

TheXtremeKing, 2021-06-26:
I've tracked it down and found a solution. 
The problem is not in this library but seems to be arduino-esp32 related.
@nseidle  maybe this will also fix your issues:
https://github.com/espressif/arduino-esp32/issues/5196

closed 2021-06-26 by TheXtremeKing