sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #54: NEO-M8U dead reckoning board can't access HNR data.

ChinguunU, 2021-08-05:
### Can't access HNR data
I have been trying to receive HNR data by running the example code Example6_getAutoHNRData.ino in debug mode however it doesn't seem to be receiving data. Can anyone give any tips on how i can fix this issue or where i can find some material to fix this issue.

### My Work Bench
I am using a nordic nrf52 development kit board. It is communicating to the dead reckoning module via I2C.
Also i think it is worth mentioning that i bought the NEO-M8U sparkfun board about a year ago.

### Behaviour after running code
[image]
[image]
[image]
[image]

PaulZC, 2021-08-05:
Hi Chinguun (@ChinguunU ),

There are a few things to investigate...

1) It is very unusual for UBX messages to be NACK'd (Negative - ACKnowledgement). They usually mean the sensor you have does not support that feature. The two NACKs are received when calling ```setAutoHNRATT``` and ```setAutoHNRINS```. Looking at the ```setAutoHNRINS```, the code sends a UBX-CFG-MSG "Set Message Rate" (Class 0x06, ID 0x01) message containing three bytes 28 2 1. 0x28 is the message class = HNR. 0x02 is the message ID = UBX-HNR-INS. 0x01 is the rate (one message per measurement cycle). So that all looks OK. Why is it being NACK'd? It suggests either your sensor does not support HNR-INS (maybe the firmware needs to be upgraded using u-center?). Or maybe that the rate is too fast, but you are only requesting 10Hz and that should be perfectly OK for the NEO-M8U.

2) When the data starts to arrive, the debug messages show a lot of "7F error"s. It is unusual to see so many of those. The data is arriving in blocks of 80 bytes. Does that match the HNR-PVT message length? Yes, it does. UBX-HNR-PVT contains 72 bytes in the payload. With the header, class, ID, length and checksum, that totals up to 80 bytes. So, it looks like the UBX-HNR-PVT data is arriving but is generating "7F errors". You can disable the 7F check by calling ```myGNSS.disableUBX7Fcheck();``` in the example, just after the ```myGNSS.begin();```.

So, please try two things for me:
Please [upgrade your NEO-M8U firmware](https://learn.sparkfun.com/tutorials/how-to-upgrade-firmware-of-a-u-blox-gnss-receiver?_ga=2.153893361.873356098.1628141012-559261113.1583443041) using u-center.
Please try calling ```myGNSS.disableUBX7Fcheck();``` after ```myGNSS.begin();```.

Thank you,
Paul

ChinguunU, 2021-08-05:
Thanks for the reply much appreciated. I'll try them out and let you know how i go.

PaulZC, 2021-08-07:
Hi @ChinguunU ,
Can you give me an update on this issue? Did upgrading the firmware correct this issue?
Best wishes,
Paul

PaulZC, 2021-08-10:
Closing through lack of feedback.
Please re-open if you need more help.
Best wishes,
Paul

closed 2021-08-10 by PaulZC