sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #59: Best practices for adding more config keys

nseidle, 2021-08-23:
The ZED-F9R introduces some additional config keys, namely the ESF range of keys (ex: CFG-MSGOUT-UBX_ESF_ALG_I2C). I suspect the ZED-F9T adds a few as well and more modules are always looming on the horizon.

* How @PaulZC did you export the PDF and then [import them in bulk](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/pull/144) with comments?
* How should we detect key diffs between PDFs to determine new adds?

PaulZC, 2021-08-24:
IIRC, I used the full version of Adobe Acrobat to export the PDF as text. It was then a couple of hours work to extract and format just the keys and their descriptions.

But, for new keys, there must be a simpler way to do this. I suspect the various pdf diff tools that are out there will fail as the differences between say the F9P and the F9T interface descriptions will be huge. However, there may be a 'quick' way to extract just the hex keys making it easier to spot any new ones with a simple diff. I shall think on this...

PaulZC, 2021-08-24:
Here's a method which works. It's not particularly quick or elegant, but it works...

1. Open the interface description in Adobe Acrobat Reader DC (the free version)
2. Do a ```File \ Save as Text...```
3. Save the file in ```Text (Accessible) (*.txt)``` format
4. Go make a cup of tea - this takes a while
5. Open the txt file in Notepad++ or another editor which supports Regular Expressions
6. The keys will have been saved as individual lines in the format: 0xnnnnnnnn space CR LF
7. So all we need to do is use a regex to delete everything else
8. Open Search \ Replace
9. Click the Search Mode - Regular Expression button
10. In the "Find what :" box enter: ```^(?!.*0x\d{8}\s\r\n).*```
11. Clear the "Replace with :" box
12. Click "Replace All"
13. You are left with just the keys - and a bunch of empty lines, some of which contain form feeds (\f)
14. Delete the empty lines (\r\n) by replacing \r\n with nothing
15. Delete the form feeds by replacing \f with nothing
16. Finally replace the remaining spaces (\s) with \r\n

Save the file. Use KDiff3 or another diff package to see the new additions.

PaulZC, 2021-08-24:
I've added the new keys for the [ZED-F9R](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/3bb7e300f33f79b0188f32d6829539b818c9e876) and the [ZED-F9T](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/9b62f307be0ea5bb58d21bc094a0cd4fb0d85382) to the release_candidate branch. I'll merge into main as soon as I've tested your Auto IMU code.

jafrado, 2021-08-24:
“strings <file.pdf>“ should get you the relevant text. strings file.doc | more is the way we used to read Microsoft word documents on Sun workstations in the mid-90’s 😀

nseidle, 2021-08-24:
Amazing. Thanks Paul! And this should serve as a 'how to' to our future selves.

closed 2021-08-24 by nseidle