sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #63: V2.0.14 Fails to compile for nRF52 MCU

beegee-tokyo, 2021-08-28:
### Subject of the issue
Compilation for nRF52 BSP fails with error.

### Your workbench
* GNSS breakout: [RAKwireless WisBlock RAK12500](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK12500/Overview/) HW version 1.0
* MCU board:
   * [WisBlock RAK4631 WisCore](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK4631/Overview/)
   * MCU & GNSS module plugged into Sensor slot of [RAKwireless WisBlock RAK5005-O](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK5005-O/Overview/)
* Powered over USB
* Compilation works with Library version V2.0.13

### Steps to reproduce
Compile any example for any nRF52 board using Adafruit nRF52 Arduino BSP.

### Expected behavior
Compilation error should not happen.

### Actual behavior
Compilation error
```log
.pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.cpp: In member function 'sfe_ublox_status_e SFE_UBLOX_GNSS::sendI2cCommand(ubxPacket*, uint16_t)':
.pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.cpp:2950:55: error: no matching function for call to 'write(uint8_t&, int)'
     _i2cPort->write(outgoingUBX->payload[startSpot], 1); // Thank you @Valrakk #61
                                                       ^
In file included from .pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.h:52:0,   
                 from .pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.cpp:43:   
C:\users\rakwireless\.platformio\packages\framework-arduinoadafruitnrf52@1.2100.201028\libraries\Wire/Wire.h:52:12: note: candidate: virtual size_t TwoWire::write(const uint8_t*, size_t) <near match>
     size_t write(const uint8_t * data, size_t quantity);
            ^~~~~
C:\users\rakwireless\.platformio\packages\framework-arduinoadafruitnrf52@1.2100.201028\libraries\Wire/Wire.h:52:12: note:   conversion of argument 1 would be ill-formed:
.pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.cpp:2950:51: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
     _i2cPort->write(outgoingUBX->payload[startSpot], 1); // Thank you @Valrakk #61
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
In file included from C:\users\rakwireless\.platformio\packages\framework-arduinoadafruitnrf52@1.2100.201028\cores\nRF5/Stream.h:26:0,     
                 from C:\users\rakwireless\.platformio\packages\framework-arduinoadafruitnrf52@1.2100.201028\cores\nRF5/HardwareSerial.h:25,
                 from C:\users\rakwireless\.platformio\packages\framework-arduinoadafruitnrf52@1.2100.201028\cores\nRF5/Arduino.h:56,      
                 from .pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.h:47,     
                 from .pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.cpp:43:   
C:\users\rakwireless\.platformio\packages\framework-arduinoadafruitnrf52@1.2100.201028\cores\nRF5/Print.h:55:12: note: candidate: size_t Print::write(const char*, size_t) <near match>
     size_t write(const char *buffer, size_t size) {
            ^~~~~
C:\users\rakwireless\.platformio\packages\framework-arduinoadafruitnrf52@1.2100.201028\cores\nRF5/Print.h:55:12: note:   conversion of argument 1 would be ill-formed:
.pio\libdeps\wiscore_rak4631\SparkFun u-blox GNSS Arduino Library\src\SparkFun_u-blox_GNSS_Arduino_Library.cpp:2950:51: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'const char*' [-fpermissive]
     _i2cPort->write(outgoingUBX->payload[startSpot], 1); // Thank you @Valrakk #61
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
```

Switching back to V2.0.13 solves the problem and compilation works without problems.

PaulZC, 2021-08-29:
Hi Bernd (@beegee-tokyo ),

Thank you. Corrected in v2.0.15.

Best wishes,
Paul

closed 2021-08-29 by PaulZC

beegee-tokyo, 2021-08-29:
@PaulZC 👍 
Thank you!