sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #66: Arduino DUE Setup

shubham9436, 2021-09-11:
### Subject of the issue
I have Arduino DUE board. I am trying to run a [simple example](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example13_PVT/Example3_AutoPVTviaUart/Example3_AutoPVTviaUart.ino) via UART. But I am not able to get the Latitude, Longitude values.

### Your workbench
* I am using SparkFun GPS-RTK-SMA Breakout - ZED-F9P and Arduino DUE.
* I have connected TX1(18, Arduino) to RX (F9P) and RX1(19, Arduino) to TX (F9P). GND and 5V pin are connected to each other. 
* I have powered Arduino DUE using type-B connected to my laptop.
* Arduino DUE have hardware serial. So, instead of using Software Serial, I am using Serial 1 as it does not have Software Serial library.

### Code
```
#include <SparkFun_u-blox_GNSS_Arduino_Library.h>
SFE_UBLOX_GNSS myGNSS;
#define baudRate 115200

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println("SparkFun u-blox Example");
  Serial1.begin(baudRate);
  myGNSS.begin(Serial1);
  myGNSS.assumeAutoPVT(true, true);

}

void loop()
{
  if (myGNSS.getPVT())
  {
    Serial.println();

    long latitude = myGNSS.getLatitude();
    Serial.print(F("Lat: "));
    Serial.print(latitude);

    long longitude = myGNSS.getLongitude();
    Serial.print(F(" Long: "));
    Serial.print(longitude);
    Serial.print(F(" (degrees * 10^-7)"));

    long altitude = myGNSS.getAltitude();
    Serial.print(F(" Alt: "));
    Serial.print(altitude);
    Serial.print(F(" (mm)"));

    byte SIV = myGNSS.getSIV();
    Serial.print(F(" SIV: "));
    Serial.print(SIV);

    Serial.println();
  }
  else
  {
    delay(5);
    static int counter = 0;
    counter++;
    if (counter > 10)
    {
      Serial.print(".");
      counter = 0;
    }
  }
}
```

### Expected behavior
It should print the latitude, longitude values.

### Actual behavior
It just prints so many dots.
Output:
```
SparkFun u-blox Example
...................................................................
```

PaulZC, 2021-09-11:
Hi @shubham9436 ,
It looks like you are attempting to use 115200 baud to communicate with the ZED-F9P. The ZED-F9P defaults to 38400 baud - as used in the original example. Please change your #define baudRate to 38400.
Please also check your antenna has a clear view of the sky.
I am going to close this issue as I believe this will correct your problem, but please re-open if you need more help.
Best wishes,
Paul

closed 2021-09-11 by PaulZC

PaulZC, 2021-09-11:
Also, please be careful with your use of assumeAutoPVT. Please look again at the original example. I think you should be using setAutoPVT?

shubham9436, 2021-09-12:
> Hi @shubham9436 ,
> It looks like you are attempting to use 115200 baud to communicate with the ZED-F9P. The ZED-F9P defaults to 38400 baud - as used in the original example. Please change your [...]

I changed the default baud rate to 115200. The key thing was the antenna having clear view of the sky. 
Also, ran the dynamic model example, and it works well

PaulZC, 2021-09-12:
Hi Shubham (@shubham9436 ),
Glad that’s working for you. Thank you for the update. 
Very best wishes,
Paul