sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #68: ZED-F9P with Portenta H7 - no i2c or UART connection

RoboCode23, 2021-09-23:
Trying to run examples for the [ZED-F9P RTK2 board](https://www.sparkfun.com/products/15136) on the Portenta H7, but can't get past myGNSS.begin() either on I2C or UART. Issue is very similar to [this](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/issues/175) or [this](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/issues/163) respectively. 
I've cut the pads to the pullup resistors, tried 100kHz I2C clock, but still no game.
Wiring is ok, since when I use the I2C scanner, it returns the correct address (0x42). I even tried changing the address or setting up the ZED-F9P with U-center so that it only sends UBX data over I2C - thought maybe the bus got flooded by NMEA sentences, but still no game.

Using latest ZED-F9P firmware(v1.13) and H7 bootloader(v22).

I have 3 pieces of the ZED-F9P and all have the same behaviour. Wierd thing is, that the examples work fine with Teensy 4.1 and what is even wierder is that when I use this code to "ping" the board in the sketch
```
Wire.beginTransmission(0x42);
Serial.println(Wire.endTransmission());
```
it works normally (prints 0), but even when I edit the "isConnected" function in the library .cpp from
```
boolean SFE_UBLOX_GNSS::isConnected(uint16_t maxWait)
{
  if (commType == COMM_TYPE_I2C)
  {
    _i2cPort->beginTransmission((uint8_t)_gpsI2Caddress);
    if (_i2cPort->endTransmission() != 0)
      return false; //Sensor did not ack
  }
```
to
```
boolean SFE_UBLOX_GNSS::isConnected(uint16_t maxWait)
{
  if (commType == COMM_TYPE_I2C)
  {
   Wire.beginTransmission(0x42);
    if (Wire.endTransmission() != 0)
      return false; //Sensor did not ack
  }
```
the issue persists and the function still returns "false".

When I connnect other sensors (mpu6050 and 9250) on the bus, they work fine. But the ZED-F9P is still in this limbo, regardless of other sensors connected or ZED being alone on the bus.

Any clues as to what could be causing this would be appreciated.

PaulZC, 2021-09-24:
Hi @RoboCode23 ,
I think the root cause here is I2C clock-stretching. The ZED uses sub-one-clock-cycle clock-stretching and we think the Portenta does not like that. We've seen the [same issue with the BNO080](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/issues/80) - which also uses clock-stretching - and we haven't got a solution for that device either.
I'm happy to discuss this further, but I really don't have a solution for you. Sorry!
Very best wishes,
Paul

PaulZC, 2021-09-24:
It should work OK on UART though. Please check you are using 38400 baud - which is the default for the ZED. Please also check your wiring: TX->RX, RX->TX, GND->GND.

RoboCode23, 2021-09-24:
Hi @PaulZC ,
thank you for the insight, abandoning effort to get it running over I2C, UART works, although Portenta datasheet/pinout says it has 4x UART, I can only get it working on 2 (Serial1 and Serial3). Serial2 doesn't even init with board not connected to anything, let alone in breakout. But that's an issue for someone else to sort out.
Thanks again for the quick response!

closed 2021-09-24 by RoboCode23