sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #87: Problem compile BasciNMEARead to UNO

kmlcmveloci, 2021-12-08:
### Subject of the issue
Problem compile BasciNMEARead to UNO

Error shown:
In file included from C:\Users\kim\Documents\Arduino -F\libraries\SparkFun_u-blox_GNSS_Arduino_Library\src/SparkFun_u-blox_GNSS_Arduino_Library.h:57:0,
                 from C:\Users\kim\Documents\Arduino -F\libraries\SparkFun_u-blox_GNSS_Arduino_Library\examples\Example1_BasicNMEARead\Example1_BasicNMEARead.ino:26:
C:\Users\kim\Documents\Arduino -F\libraries\SparkFun_u-blox_GNSS_Arduino_Library\src/u-blox_structs.h:1830:53: error: size of array 'data' is too large
   UBX_MGA_DBD_data_t data[UBX_MGA_DBD_RINGBUFFER_LEN]; // Create a storage array for the MGA DBD packets

PaulZC, 2021-12-08:
Hi @kmlcmveloci ,

Please upgrade to v2.1.1 of the library. Does that fix this issue?

If not, please tell me which board you are using (manufacturer and type).

Thank you,
Paul

kmlcmveloci, 2021-12-08:
It is 2.1.1, but the board was set to a uno with ethernet, and this was not defined in the u-blox_structs.h line: 1820! Only UNO board is defined to reduce ram!

PaulZC, 2021-12-08:
Hi @kmlcmveloci ,

Thank you for reporting this. I have corrected it in v2.1.2.

Best wishes,
Paul

closed 2021-12-08 by PaulZC