sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #91: Problems with Ex14 NTRIP Caster Unauthorized

nseidle, 2021-12-12:
Moved from issue [83](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/83).

Problem getting ex 14 working with rtk2go:

[image]

Unauthorized access error from RTK2go.

nseidle, 2021-12-12:
The 'SparkFun_RTK_Surveyor' is a hold over from the RTK project and will be corrected.

I cannot replicate the problem right now.

[image]

Using our dummy credentials can you connect successfully?

nseidle, 2021-12-12:
Perhaps you're using a caster that does not allow open casters. SNIP has a [good explanation](https://www.use-snip.com/kb/knowledge-base/pre-ntrip-client-connections/). Base64 would then indeed be needed. Let me see if I can replicate your error on Emlid (closed casting).

jkreienh, 2021-12-13:
@nseidle I think I've got it working.  Somehow I got the Pw and MtPnt backward during all my troubleshooting. 

http://rtk2go.com:2101/SNIP::MOUNTPT?NAME=AUS_TX

nseidle, 2021-12-13:
Good to know! Glad you got it working.

closed 2021-12-13 by nseidle