sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #94: UBX_Integrity_Checker : incorrect error detection

Eric-FR, 2021-12-21:
UBX_Integrity_Checker is reporting a lot of errors on my ubx log files recorded since 21/12/17.

Sparkfun RTK Express with  [RTK_Surveyor_Firmware_v110-RCDec-13.zip](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/7707467/RTK_Surveyor_Firmware_v110-RCDec-13.zip). Logging on the µSD card.

Static recording with or without RTK correction.

UBX_Integrity_Checker is complaining about:
`Panic!! Excessive NMEA message length!`

[image]

[image]

Sample log (without RTK correction):
[SFE_Express_211221_122914.zip](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/files/7757937/SFE_Express_211221_122914.zip)

No problem with [ubx_convert](https://github.com/Stefal/ubx-to-gpx), u-center or rtklib (RTKPlot and RTKConv).

RTK Express settings: 
[SFE_Express_Settings.txt](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/files/7758001/SFE_Express_Settings.txt)

Eric-FR, 2021-12-21:
With ubx_convert:
[image]

PaulZC, 2021-12-22:
Hi @Eric-FR ,

Thank you for reporting this.

The code was using  a maximum NMEA length ( ```max_nmea_len``` ) of 100 characters:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/0b298e7c6712ef165c9f68bb3ccc5e0b5ade1b90/Utils/UBX_Integrity_Checker.py#L149

But, from your data, I can see that the **PUBX** (u-blox proprietary NMEA messages) can exceed that.

I have updated the tool: I have increased the maximum length to 128; and also record the longest NMEA message length.

[image]

Enjoy!
Paul

closed 2021-12-22 by PaulZC

Eric-FR, 2021-12-28:
Do you have any reason to believe that NMEA sentences length will not be higher than 128?
I just found one with 116 characters:

`$PUBX,00,122623.00,4511.08102,N,00544.93346,E,264.521,D3,0.014,0.012,0.013,0.00,-0.006,1.0,0.58,0.86,0.60,25,0,0*55`

PaulZC, 2021-12-28:
128 is just an "educated guess"!

If you see PUBX messages longer than this, please let me know. Thanks!

Eric-FR, 2021-12-30:
Although I can't find the official source of NMEA standard (that may not be freely available), it is commonly stated like in [wikipedia](https://en.wikipedia.org/wiki/NMEA_0183) that:

_Messages have a maximum length of 82 characters, including the $ or ! starting character and the ending <LF>_

So, PUBX don't comply with NMEA standard, isn't it?

Should we have a soft warning for sentences exceeding 82 characters and stronger action over a higher limit like 128 characters?

Eric-FR, 2021-12-30:
I just found classical sentences exceeding the official limit:

`$GNGGA,185118.00,4533.5955918,N,00555.8326327,E,4,12,0.68,316.817,M,47.337,M,1.0,0000*6E`

nseidle, 2021-12-30:
Is that with high precision turned on?