sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #97: UBX_NAV_PVAT not supported

Cils, 2022-01-11:
### Subject of the issue
The messages UBX_NAV_PVAT are not supported.

### Your workbench
* ZED-F9R, with newest firmware (v. 1.21) 
* Here the rest but probably not relevant
* ESP32
* Qwiic
* USB

### Steps to reproduce
Try to use the UBX_NAV_PVAT output messages.

### Expected behavior
Use of messages of type UBX_NAV_PVAT. E.g.: message pull and log, ... 

### Actual behavior
The message type UBX_NAV_PVAT  are not implemented in the library.

PaulZC, 2022-01-14:
Hi @Cils,

Thank you for raising this issue. We have added full support for PVAT in v2.2.0 of the library. It will be released in a few minutes.

Best wishes,
Paul

PaulZC, 2022-02-05:
Hi @Cils ,

Just to let you know: I found a bug in the "auto" support for PVAT. I had forgotten to include PVAT in ```getMaxPayloadSize```. Polling with ```getNAVPVAT`` will work, but the full "auto" support (including data logging) will fail.

I will release a fix for this in a few days. I have corrected it - but I cannot release it yet as I am in the middle of making other changes to the library.

Best wishes,
Paul

Cils, 2022-02-10:
Many thanks for adding the PVAT support and letting me know of the ongoing problem. I noticed that the "auto" PVAT was not working but didn't had time to look into it. You saved me some troubleshooting time. 

best wishes,
Cils

PaulZC, 2022-02-11:
Reopening this, just in case any other users come across the same issue.
It will be resolved in the next release.
For now, please use the release_candidate branch if you need full "auto" PVAT.

closed 2022-02-20 by PaulZC