sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #100: Possible :) bug in use of UBX-CFG-NAV5 within SFE_UBLOX_GNSS::setDynamicModel()

D2We, 2022-01-17:
### Subject of the issue
Hello. This is my first use of GitHub, so I hope this is the correct means to report a possible issue with this library.
I was extending the library to give me access to the Static Hold Threshold, and I was copying the existing code from SFE_UBLOX_GNSS::setDynamicModel() as it also uses UBX-CFG-NAV5. I noticed in the code that once the existing navigational models were read, the mask at offset 0 was assigned 1 (to select the application of the dynamic model settings).

payloadCfg[0] = 0x01;            // mask: set only the dyn bit (0)

I believe there is a possible issue in that any other bits in this mask (and their associated mechanisms) will be disabled.

According to my documentation (u-blox M8 UBX-13003221 - R26), the default setting for this mask has many bits set (some which it states are recommended NOT to change).

I would think that this code is what is intended (OR-ASSIGN):

payloadCfg[0] |= 0x01;           // mask: set only the dyn bit (0)

As I am new to the device, I do not yet have the knowledge to determine how (or if) this change effects the device. From my (limited) searches through the code, only SFE_UBLOX_GNSS::setDynamicModel() has this possible issue.

### Your workbench
I'm using an Arduino MKR GPS shield (with u-blox SAM M8 GNSS) and this Sparkfun library version 2.1.15.

PaulZC, 2022-01-17:
Hi @D2We ,

This is a subtle one...

From the interface description:

[image]

The wording is really specific: "Only the masked parameters will be applied.".

So it is OK to set ```payloadCfg[0] = 0x01; // mask: set only the dyn bit (0)``` because we _only_ want to _change_ the dynamic model setting. We want all the others to remain the same.

For your Static Hold Threshold function, you're going to want to use:

```payloadCfg[0] = (1 << 6); // mask: set only the staticHoldMask bit (6)``` .

Best wishes,
Paul

closed 2022-01-17 by PaulZC

gabrielesartor, 2022-05-03:
Hi @D2We!
I'm trying to use the Arduino MKR GPS shield as well, but I don't know how to connect to it.
Can you tell me which interface you used? Thank you!