sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #101: Using this library with UART without Arduino dependency?

sukesh-ak, 2022-01-18:
### Subject of the issue
What are the changes required to make this library work with UART without Arduino dependency
Would like to use it with ESP-IDF based projects on ESP32

### Your workbench
ESP32

### Expected behavior
Use this library with ESP-IDF

### Actual behavior
Tell us what happens instead

PaulZC, 2022-01-18:
Hi Sukesh (@sukesh-ak ),

Sorry - you will need to figure that out for yourself. I have no experience with ESP-IDF.

The code is all written in standard C++. It should hopefully be as simple as replacing ```hwAvailable``` , ```hwPrint``` , ```hwWrite``` etc. with the ESP-IDF equivalent?

Best wishes,
Paul

closed 2022-01-18 by PaulZC

sukesh-ak, 2022-06-14:
hi @PaulZC 

Thought of sharing here, how this library can be used with ESP-IDF with detailed steps.
Write-up used this library as an example.
https://github.com/sukesh-ak/Arduino-Library-with-IDF

Please feel free to add this link to your readme if its useful.

PaulZC, 2022-06-14:
Hi Sukesh (@sukesh-ak ),

This is great stuff - thank you!

I will pin this issue - so others can find it.

Thanks again,
Paul

sukesh-ak, 2022-06-14:
Thank you for this amazing library @PaulZC  👍

dizcza, 2024-03-04:
I've successfully built this lib using [Arduino as an ESP-IDF component[](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html#arduino-as-an-esp-idf-component)](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html) with the following cmake:

```
file (GLOB SOURCES *.cpp)
idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS "."
                       REQUIRES arduino)
target_compile_options(${COMPONENT_LIB} PRIVATE -std=gnu++11)
```

I can simply add this simple fine (need to modify the paths) to support ESP-IDF.

@sukesh-ak didn't about such a lib, thanks. Have you tried its I2C API?

sukesh-ak, 2024-03-05:
> I've successfully built this lib using [Arduino as an ESP-IDF component](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html) with the following [...]

This was done few years ago.
I am working to avoid dependency with `esp32-arduino core` completely now. Will share once I get time to test it.

I don't have the hardware with I2C, so not tried yet. Maybe will try in near future with one of my custom hardware.