sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #102: ZED F9P - Unable to open the port

niranjanravilevelup, 2022-01-19:
### Subject of the issue
I am working on integrating ZED F9P with NVIDIA Jetson NX device via USB connection. 

First, I tried with python library using the below code: https://github.com/sparkfun/Qwiic_Ublox_Gps_Py/blob/master/examples/geo_coords_ex1.py

Replaced port = serial.Serial('/dev/serial0', baudrate=38400, timeout=1) with  port = serial.Serial('/dev/ttyACM0', baudrate=38400, timeout=1). 
It worked fine but the GPS was unable to lock signals at all times.  

Next step, was trying to use C libraries to verify the same issue :  https://github.com/sugbuv/Ublox_Linux_Library/blob/master/linux_examples/ublox_f9p_test.cpp 

But I get an error stating "Ublox is not connected. Please connect ublox GNSS module and try again..." 

It worked with python libraries but not with C. It would be helpful to get inputs on this issue or any workarounds. 

Hardware:
ZED F9P with GNSS Antenna
NVIDIA Jetson NX
USB Connection

niranjanravilevelup, 2022-01-19:
Got it fixed, it was an issue with sudo access.

closed 2022-01-19 by niranjanravilevelup

PaulZC, 2022-01-19:
Thanks for the update. Glad it’s working for you. There is more information about the sugbuv code in issue #65 .

Best wishes,
Paul

niranjanravilevelup, 2022-01-19:
@PaulZC  Thank you. I have an additional question. When i am testing the GPS module , it doesn't get locked with satellites. Most of times I see null values for lat/long and other parameters. 

i tried in python, c and u-center software. Still cant seem to fix the issue. Would be helpful to get your inputs on this.

PaulZC, 2022-01-19:
Make sure your antenna has a clear view of the sky!

Which antenna are you using?

niranjanravilevelup, 2022-01-19:
Yes, it is having a clear view of the sky.  Antenna Link: Link: https://www.u-blox.com/en/product/ann-mb-series

Still it does not pick up.

PaulZC, 2022-01-19:
That antenna is a very good choice.

Do you have access to a Windows computer? u-blox u-center will show how many satellites are being tracked.

PaulZC, 2022-01-19:
Also have a look at the antenna SMA connector. Make sure the center pin has not been pushed back into the cable. It has happened to me more than once!

Best wishes,
Paul