sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #113: Example14_NTRIPServer

zlotyGNSS, 2022-02-24:
Hello, I am the happy owner of the GPS-RTK2 Board-ZED-F9P and SparkFun Thing Plus - ESP32 WROOM modules. I combined them with a simple and great Qwiic Connect System. I made one of the simplest base stations you described in the base station manual. Such miracles can of course be done thanks to your Arduino library for u-blox modules, I used the code with
SparkFun_u-blox_GNSS_Arduino_Library / examples / ZED-F9P / Example14_NTRIPServer. Everything works beautifully, the corrections reach the rover, and it measures every centimeter, I checked around the house and 10 km from it :). I only have a question whether (example 14) in the Arduino library can be somehow modified so that the modules work outside the computer with the serial port screen in the Arduino. The point is that the module starts sending corrections only after typing a character or letter to the serial port (screen), and it would be nice if it all worked without a computer, i.e. regardless of the letter entry, without a computer. Can this Arduino code be slightly modified to let the software run automatically without user interaction? Such a solution would allow the use of this base station, among others in coordinate points, somewhere outside the house in the field. I tried to change the code, but without effective, I think I have too little programming knowledge, please help.

PaulZC, 2022-02-25:
Hi @zlotyGNSS ,

Please try this:

Replace these two lines:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/b75869936046dc36422bdd5678e5eee94350e44b/examples/ZED-F9P/Example14_NTRIPServer/Example14_NTRIPServer.ino#L148-L149

with the following:

```
  static bool firstTime = true;
  if ((Serial.available()) || firstTime)
  {
    firstTime = false;    
    beginServing();
  }
```

Please test this and, if it does what you need, please "Close with comment".

Best wishes,
Paul

zlotyGNSS, 2022-02-25:
Thank you so much for your help Paul :) for sharing your great Arduino programming knowledge. 
Everything works now as I dreamed. The Esp32 module in tandem with RTK2 now automatically connects to the EMLID caster without user interaction, and at the same time it is possible to stop sending NTRIP corrections at any time with a sign. The base station can work on a powerbank, independently of the computer, on points with known coordinates, for example on a geodetic network. 
Thank you again and best regards from Poland. 
Charles.

closed 2022-02-25 by zlotyGNSS

PaulZC, 2022-02-25:
Hi Charles (@zlotyGNSS ),
You are very welcome. I am glad that is working for you.
Best wishes,
Paul