sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #121: The UBX-MON-RF (0x0a 0x38)  function is not supported by the current SparkFun_u_blox_GNSS_Arduino_Library-2.1.4  And i have written the various parts of code and added them to the current code.  i am not interested in knowing if there is a way it can be merged into the latest code going forward? i can upload the various files

gbittner99, 2022-03-01:
Ublox MONRF not supported in current 2.2.14 version of code

https://www.u-blox.com/en/docs/UBX-21022436 page 127  3.14.10 
UBX-MON-RF (0x0a 0x38)   

working with Arduino MEGA 2560 with Arduino IDE 1.819
using Sparkfun GPS 15733 module connected on Serial port 1 of arduino
all powered by USB and all working showing GPS coordinates and good signal strengths.

As the function was not included in the current code I have implemented the code to pick up The MonRF function and have a working hardware and a sketch picking up the data and displaying it on a TFT touchscreen display.

As the software is not included in the standard libraries I would like to get it included but am not familiar with the details on how to do this.  Also i am not sure that i have followed all of the correct formatting with my additional code. 
below is screenshot from TFT diaplay of working unit showing values from the MON-RF function and other functions.

[image]

PaulZC, 2022-03-02:
Hello @gbittner99 ,

I can see how being able to monitor the jamming / interference level could be very useful. Especially in certain parts of Eastern Europe at the moment.

I have added support for UBX-MON-RF and UBX-MON-HW so you can check the jamming status on M8 modules as well as ZED-F9. Please see the new Example29 for more details.

v2.2.3 will be released in a few minutes.

Good luck with your project.

Best wishes,
Paul

closed 2022-03-02 by PaulZC

gbittner99, 2022-03-02:
Thank you very much for your Quick implementation of this.. We in Canada are not in the Situation that is in eastern Europe. We have experienced situations with vehicles with multiple GPS antennas creating interference between the antennas. the solution was to remove one or more of the antennas and use a splitter to run the multiple GPS devices but it took months to find the issue also we service a lot of GPS devices in vehicles So we have created a handheld device to allow us to test the antennas performance and not have to do the labor of changing out a $300 Sharkfin antenna to find out that it was not the issue. The older antennas we used could be unscrewed and a new one tested in a few minutes from the outside of the vehicle without having to remove parts of the interior of the vehicle to just test it.. Again thank you very much for implementing this function. should more of these be built after i retire in a month my co-workers do not have to maintain a custom library to keep it current/working. (currently building 3 more of them for our service trucks, etc) 

Closed [ https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/121 | #121 ] via [ https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/pull/122 | #122 ] .