sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #125: Wrong Leap Year Calculation

tim-scherer, 2022-03-04:
### Subject of the issue
- The leap year calculation yields that 2022 is a leap year; but next leap year is 2024. 
- As a result of this, epoch time calculation error is 86400 s (24 h) wrong. 
- 
### Your workbench
SparkFun_u-blox_GNSS_Arduino_Library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library Library V2.2.0

### Steps to reproduce
The error is located in the calculation perfomed in
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/6eea8776152233468b9883981a551875a0a50ca0/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L16124
and
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/6eea8776152233468b9883981a551875a0a50ca0/src/SparkFun_u-blox_GNSS_Arduino_Library.h#L625

### Expected behavior
2022 is not a leap year

### Actual behavior
DAYS_SINCE_MONTH[((uint32_t)packetUBXNAVPVT->data.year - 1970) & 3][(uint32_t)packetUBXNAVPVT->data.month] results as a leap year

PaulZC, 2022-03-04:
Hi @tim-scherer ,

Wow. Thanks for reporting this! That gremlin has been in there for a long time... But I guess only appeared on Tuesday...

We'll fix this in the next release.

Best wishes,
Paul

tim-scherer, 2022-03-04:
> Hi @tim-scherer ,
> 
> Wow. Thanks for reporting this! That gremlin has been in there for a long time... But I guess only appeared on Tuesday...
> 
> We'll fix this in the next release.
> 
> Best wishes, Paul

This could be a fix, changing order of array (idx 0 to 3):
[image]

PaulZC, 2022-03-04:
The corrections are in the [release_candidate branch](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/release_candidate) if you would like to test them?

Please see [this commit](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/commit/59d15e014f006e6551a85e1592f127d6b9a5b703) for the changes. (I updated the examples too.)

papillon81, 2022-03-06:
Confirmed working. Noticed the same 24h delta today. :)

PaulZC, 2022-03-07:
Hi @tim-scherer / @papillon81 ,

Thank you both. I will release v2.2.5 in a few minutes.

Best wishes,
Paul

closed 2022-03-07 by PaulZC