sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #127: Does this support ESP32?

tmargot, 2022-03-08:
### Subject of the issue
Describe your issue here. If you reference a datasheet please specify which one and in which section (ie, the protocol manual, section 5.1.2). Additionally, screenshots are easy to paste into github.

### Your workbench
* What development board or microcontroller are you using?
* What version of hardware or breakout board are you using? 
* How is the breakout board wired to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

PaulZC, 2022-03-09:
Hello @tmargot ,

Yes, it works well on ESP32.

Best wishes,
Paul

closed 2022-03-09 by PaulZC

tmargot, 2022-03-09:
> <sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/127/1062694286@
> github.com>

Does one have to swap out SoftwareSerial for Hardware serial? I tried and
got a panic and ESP32 just kept rebooting.

PaulZC, 2022-03-09:
Which example are you trying to run?

PaulZC, 2022-03-09:
Try [Example12](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example12_UseUart/Example12_UseUart.ino). It works for me on a SparkFun ESP32 Thing Plus with both hardware serial and software serial:

```
#include <SoftwareSerial.h>
SoftwareSerial mySerial(16, 17); // RX, TX

//#define mySerial Serial1 // Hardware Serial
```

```
//#include <SoftwareSerial.h>
//SoftwareSerial mySerial(16, 17); // RX, TX

#define mySerial Serial1 // Hardware Serial
```

tmargot, 2022-03-09:
> <sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/127/1062879437@

Hi,

Thank you for the reply, yes I tried Example 12

Top of code:
#include <SparkFun_u-blox_GNSS_Arduino_Library.h> //
http://librarymanager/All#SparkFun_u-blox_GNSS
SFE_UBLOX_GNSS myGNSS;

#include <SoftwareSerial.h>
SoftwareSerial mySerial(16, 17); // RX, TX

//#define mySerial Serial1 // Hardware Serial
//#include <SoftwareSerial.h>
//SoftwareSerial mySerial(16, 17); // RX, TX

#define mySerial Serial1 // Hardware Serial

long lastTime = 0; //Simple local timer. Limits amount of I2C traffic to
u-blox module.

Serial  Monitor Output:

18:55:49.808 -> ets Jun  8 2016 00:22:57
18:55:49.808 ->
18:55:49.808 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:55:49.808 -> configsip: 0, SPIWP:0xee
18:55:49.808 ->
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:55:49.808 -> mode:DIO, clock div:1
18:55:49.808 -> load:0x3fff0018,len:4
18:55:49.855 -> load:0x3fff001c,len:1216
18:55:49.855 -> ho 0 tail 12 room 4
18:55:49.855 -> load:0x40078000,len:10944
18:55:49.855 -> load:0x40080400,len:6388
18:55:49.855 -> entry 0x400806b4
18:55:49.948 -> SparkFun u-blox Example
18:55:49.948 -> GNSS: trying 38400 baud
18:55:49.948 -> Guru Meditation Error: Core  1 panic'ed
(IllegalInstruction). Exception was unhandled.
18:55:49.948 -> Memory dump at 0x400d4e60: dc574c74 d757dd54 64c455dc
18:55:49.948 -> Core 1 register dump:
18:55:49.995 -> PC      : 0x400d4e64  PS      : 0x00060530  A0      :
0x800d7365  A1      : 0x3ffb1f80
18:55:49.995 -> A2      : 0x3ffc0580  A3      : 0x3ffbff30  A4      :
0x3ffc0564  A5      : 0x80000020
18:55:49.995 -> A6      : 0x0800001c  A7      : 0x00004e20  A8      :
0x800d109a  A9      : 0x3ffb1f40
18:55:49.995 -> A10     : 0x3ffbff30  A11     : 0x3ffc0564  A12     :
0x0000044c  A13     : 0x00000000
18:55:49.995 -> A14     : 0x00000009  A15     : 0x00000000  SAR     :
0x00000016  EXCCAUSE: 0x00000000
18:55:49.995 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    :
0x4000150d  LCOUNT  : 0xffffffff
18:55:50.041 ->
18:55:50.041 -> ELF file SHA256: 0000000000000000
18:55:50.041 ->
18:55:50.041 -> Backtrace: 0x400d4e64:0x3ffb1f80 0x400d7362:0x3ffb1fb0
0x40086d8d:0x3ffb1fd0
18:55:50.041 ->
18:55:50.041 -> Rebooting...

I am using one of these:
https://raw.githubusercontent.com/AchimPieters/esp32-homekit-camera/master/Images/ESP32-30PIN-DEVBOARD.png

Any ideas?

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

tmargot, 2022-03-09:
Hi Paul,

Noticed that I had ESPSoftwareSerial installed. I have removed it and now I
get error message that I do not have SoftwareSerial.h. As another attempt,
I followed the install instructions for the board mentioned here
https://learn.sparkfun.com/tutorials/esp32-thing-plus-hookup-guide/software-setup,
but still no luck nor SoftwareSerial.h.

After all this I have got it to work with HardwareSerial, I think
referencing Serial2 was the answer for RX2/TX2. Many thanks for you help,
much appreciated.

Code:

#define mySerial *Serial2* // Hardware Serial
#define RXD2 16
#define TXD2 17

long lastTime = 0; //Simple local timer. Limits amount of I2C traffic to
u-blox module.

void setup()
{
  Serial.begin(115200);
  while (!Serial); //Wait for user to open terminal
  Serial.println("SparkFun u-blox Example");

  //Assume that the U-Blox GNSS is running at 9600 baud (the default) or at
38400 baud.
  //Loop until we're in sync and then ensure it's at 38400 baud.
  do {
    Serial.println("GNSS: trying 38400 baud");
    //mySerial.begin(38400);
    *mySerial.begin(38400, SERIAL_8N1, RXD2, TXD2);*

Thank you,
Tristan

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>