sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #132: I2C NACK issue for ZED-F9P

arnaudcazale, 2022-04-05:
### Subject of the issue
I2C NACK low driver error

### Your workbench
* dev board= Nordic nrf528332
* breakout board = GPS-RTK2 (ZED-F9P)
* wired with 15cm simple cables (VCC/GND/SDA/SCL)
* dev board is powered from USB
* Everything works well with SAM-M8Q
### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Device should handle data and respond

### Actual behavior
Every time i try to send data via I2C, i have low driver internal I2C NAK error. It seems the device don't handle the bus for acknowledgement.

PaulZC, 2022-04-05:
Hi @arnaudcazale ,

It is going to be difficult for me to replicate this issue. I do not have any nRF528332 hardware to test...

So, just to be clear, you are saying that I2C communication with the SAM-M8Q works OK? You only see these NACKs when using the ZED-F9P?

This is very surprising because the I2C interface on the SAM and the ZED is identical - as far as I am aware.

Do you have the pull-up resistors configured the same on both boards?

Can you please tell us which code example you are using?

Are you seeing I2C Bus NACKs? Or are you seeing UBX protocol NACKs?

Can you share any logic analyzer data which shows the issue? And/or please share the debug messages from the library (please call ```myGNSS.enableDebugging();``` _before_ ```myGNSS.begin();```).

Best wishes,
Paul

PaulZC, 2022-04-06:
Hi @arnaudcazale ,

One thing to check... Has the ZED's I2C address been changed? That would explain the NACKs (on address 0x42).

Please try running [Example9](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example9_ChangeI2CAddress/Example9_ChangeI2CAddress.ino).

Best wishes,
Paul

arnaudcazale, 2022-04-06:
Hi @PaulZC , thanks for your reply.
Yes, I2C communication works for SAM-M8Q, but not for ZED-F9P. 
I talk about NACK on I2C low level driver, not NACK of UBX protocol. 
For being precise, i rewrite a complete solution for Nordic target (works with SAM-M8Q), based on your library, so i can't run your example "out of the box".

As an example, if i try to get the frequency, i write [ 0xB5 0x62 0x06 0x08 0x00 0x00 0x0E 0x30 ] to the address 0x42.
You can see on the pic below the I2C signal for SAM-M8Q (between 8th and 9th clock, device behave differently: SAM is ACK and ZED is NACK)
[image]

And here for the ZED-F9P
[image]

I don't think it's a pull-up or CLK speed issue (doesn't work if a disable on board pull-up or is the speed varies from 100KHz to 400KHz, behavior is the same).

Your issue coming from address is interesting, i will do an I2C scan, but i try the board out of the box, and i'm sure never has changed the default address. IS it possible to have a device with different defaut address ?

PaulZC, 2022-04-06:
Hi @arnaudcazale ,

Very interesting!

Have you tried using u-center? You can install u-center on a Windows PC and connect to the GPS-RTK2 via USB. It will allow you to read the I2C address, and also monitor the I2C bus traffic. You can have USB and I2C connected at the same time. UART too.

https://www.u-blox.com/en/product/u-center

Download u-center - not u-center 2. u-center 2 is very different.

When you are connected, open the View \ Messages View window. Select UBX\CFG\PRT and change target to 0-I2C. Click "Poll" at the bottom of the screen if required. You should see the address 0x42:

[image]

UBX-MON-COMMS will show if any I2C messages have been sent or received. I2C is missing in my screenshot - I only have USB connected right now:

[image]

Let me know what you find.

Best wishes,
Paul

arnaudcazale, 2022-04-06:
Hi @PaulZC 

I have connected the ZED with u-center, made an I2C scan an see 0x42. 
After that, i try to run my FW, and it was working...!

I realize that it works only with USB plugged into the board. I've gotten use to deal with the SAM-M8Q, which haven't USB powering utilities. 

In fact, before, i was testing just by powering througt 3V3 (input near SDA connector). Sorry for the wrontg issue description. 

I have to check the hardware schematics too see if i can power the GPS only from the 3V3. DO you have some advice about that ? (I really need the board being embedded)

PaulZC, 2022-04-06:
Hi Arnaud (@arnaudcazale ),

I am pleased the board is working for you.

I (almost) always connect my ZED board using only the 4-pin Qwiic connection (3.3V, 0V, SDA, SCL). It works well. USB power is helpful, but not essential.

The ZED can draw as much as 130mA (at 3.3V). Even more if you have an active antenna connected (perhaps another 20mA to 40mA depending on the antenna). Maybe that is too much for your nRF board?

For comparison, the SAM draws 32mA (typical), 67mA (peak).

I am going to close this issue, but please reopen it if you need more help.

Very best wishes,
Paul

closed 2022-04-06 by PaulZC