sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #140: Library for MKR GPS Shield

gabrielesartor, 2022-05-15:
### Subject of the issue
Hi, I'm trying to use your library to work with the MKR GPS Shield of Arduino. It seems that this library should work according to this page https://support.arduino.cc/hc/en-us/articles/360018628960-The-MKR-GPS-shield-does-not-fetch-any-data. Probably it could work because the shield use the SAM-M8Q which you are able to manage.
I've seen that your library give an example of how to connect with UART and I2C, do you know how to connect "as shield". I can't connect it in other ways.
I know that it's not your own product, but I'd appreciate if you let me know if you've already solved this problem or you have any advice.
Thank you in advance!

### Your workbench
- Arduino MKR WAN 1310 (powered via USB)
* Arduino MKR GPS Shield (connected as shield)

PaulZC, 2022-05-15:
Hi Gabriele (@gabrielesartor ),

It looks like our library should work without any modifications. It looks like you can use both serial and I2C. For I2C, you will need a 5-pin to 5-pin sensor cable to connect the shield to the board. I can not  find the name of the MKR WAN's serial port (TX and RX pins). But if it is "Serial1", then you can use (e.g.) Example12:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example12_UseUart/Example12_UseUart.ino

Replace these lines:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/5989aecee84066d27db253a055a608b3ce7d7df2/examples/Example12_UseUart/Example12_UseUart.ino#L33-L34

With:

```
#define mySerial Serial1
```

Let me know if this works.

Best wishes,
Paul

gabrielesartor, 2022-05-15:
Hi Paul, thank you for your prompt response!
Actually, I've tried that solution but it seems that the library SoftwareSerial.h is not available for my board (MKR WAN 1310).
There is an alternative library to do it?

PaulZC, 2022-05-15:
I found these pin definitions in the mkrwan1300 variant.h. It looks like "Serial1" should work OK?

[image]

Do not use SoftwareSerial. Delete those two lines. Use ```#define mySerial Serial1``` and it should work?

Best wishes,
Paul

gabrielesartor, 2022-05-15:
Something seems working!
Thank you so much!

closed 2022-05-15 by gabrielesartor

gabrielesartor, 2022-05-15:
@PaulZC last question: is this output normal?
[image]
I mean, why there are no commas in latitude and longitude? There are other print formats?

PaulZC, 2022-05-15:
Hi Gabriele (@gabrielesartor ),

Yes, perfectly normal. The values are degrees * 10^-7. Multiply by 0.0000001 to convert to degrees. Your location is 45.1 degrees North, 7.7 degrees East.

I hope the weather is nice in Turin! ;-)

Best wishes,
Paul

gabrielesartor, 2022-05-15:
Ok, perfect!
Thank you for editing the comment :D