sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #154: Add SparkFun_u-blox_GNSS_Arduino_Library library to Arduino IDE 2.0 "library manager"

jh83, 2022-09-15:
### Subject of the issue
When searching for the "SparkFun_u-blox_GNSS_Arduino_Library" in the new Arduino IDE 2.0, only the old "SparkFunk u-blox arduino library" version 1.8.11 is returned.

Can you help with adding the SparkFun_u-blox_GNSS_Arduino_Library library so that it is visible in the Arduino IDE 2.0 library manager?

### Your workbench
Windows 10

### Steps to reproduce
Install Arduino IDE 2.0.
Go to library manager and search for SparkFun_u-blox_GNSS_Arduino_Library
One result is shown, but this result is the old version and not the up-to-date package.

### Expected behavior
Same results as in the "old" Arduino IDE in which the latest version of the SparkFun_u-blox_GNSS_Arduino_Library is visible

### Actual behavior
An old version of the SparkFun_u-blox_GNSS_Arduino_Library library is displayed.
[image]

PaulZC, 2022-09-15:
Hi @jh83 ,

Thanks for reporting this.

It looks like the library manager search tool is slightly different in v2.

If I search for "SparkFun u-blox GNSS" (with spaces) I see both the old and new libraries:

[image]

If I search for "SparkFun_u-blox_GNSS" (with underscores) I only see the old library:

[image]

What's more worrying is this:

With Arduino IDE 1.8, you could include hyperlinks in examples which would automatically open the library manager:

```
//Click here to get the library: http://librarymanager/All#SparkFun_u-blox_GNSS
```

With v2, those links open in a web browser and fail as they are not correctly formatted HTTP...:

[image]

OK. Thanks again for reporting. The answer to your particular issue is to search using spaces, not underscores. But it is clear we have other issues to solve with v2...

Best wishes,
Paul

PaulZC, 2022-09-15:
Submitted as an issue here:

https://github.com/arduino/arduino-ide/issues/1442

PaulZC, 2022-10-17:
Hi @jh83 ,

Arduino have fixed this issue. Please see the above link for full details. They have not yet released the changes, but it is fixed in the most recent "nightly build":

https://www.arduino.cc/en/software

Thanks again for the heads-up.

Closing...

Best wishes,
Paul

closed 2022-10-17 by PaulZC