sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #155: Some extra function

jan5650, 2022-09-21:
Hi Guys!

You did great job with this libraray. But sadly some function missed, that i realy want to use.

It would be great, when there will be a function, wich i can measure Distance and Bearingto between two gps coordinate. 

Can you write a function for it? NEOGps libraray got it, but sadly i cannot implement it by myself

Thank you for your work!

PaulZC, 2022-09-21:
Hi @jan5650 ,

I do not want to add the code for calculating the "Great Circle Distance" directly into the library as it takes up a significant amount of memory. However, I have added a [new example](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example31_Great_Circle_Distance/Example31_Great_Circle_Distance.ino) showing how you can calculate the distance and course between two points. Please see the [v2.2.16 release notes](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/releases/tag/v2.2.16) for more details.

Best wishes,
Paul

closed 2022-09-21 by PaulZC

jan5650, 2022-09-26:
**Thank you for the example!**

Can i do a last question?

In this library there is a lot of "Heading" type value. What's the different between them?

`getATTheading, getHeading, getHeadingAccEst, getVehicleHeading, getMotionHeading, getHNRheading`

I want the heading to where my vechile moving. Because there is no Compass, i can only compute heading by two GPS coordinate. Wich one i need to use fot the best resoult, or is there any meethod?

Thank you!

jan5650, 2022-10-14:
**Thank you very much, you helped me a lot!**

I have one more question: 
I would like to check somehow if there is GPS fixation, because if there are a sufficient number of satellites, I only start certain calculations when I have a sufficiently accurate latitude-longitude data. As long as there are only 0-5 satellites fixation I don't even start calculations because it is so inaccurate.

So far I have related this to the number of satellites. There was a value in the NEOGps Library that gave back the number of satelites of fixation. 

In this libraries how can I get the number of satellites. I wonder if there is a better way? Maybe there is another methode?

PaulZC, 2022-10-14:
Hi @jan5650 ,

```getFixType()``` will tell you if you have a 3D fix (```getFixType() == 3```).

```getSIV()``` will tell you the number of satellites used in the navigation solution. SIV means "Satellites In View" - an old term - but the number returned is actually the number of satellites used in the navigation solution.

```getHorizontalAccEst()``` returns the horizontal accuracy estimate in mm. If you wait until ```getHorizontalAccEst() < 3000``` you know you have a good fix.

Best wishes,
Paul

jan5650, 2022-10-14:
Wow! You are so on Top!

**If I set:**

// Turning off NMEA on both output
myGNSS.setUART1Output(COM_TYPE_UBX);
myGNSS.setI2COutput(COM_TYPE_UBX);

myGNSS.setMeasurementRate(200)
myGNSS.setNavigationRate(1)
myGNSS.saveConfiguration();

Then i get a module with 5Hz, that gives output from current latitude and longtitude in every 200ms. And then i need to call myGNSS.getLatitude(); and myGNSS.getLongitude(); in every loop after 200ms?

Am i right?