sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #158: NEO-D9C with ZED F9P

lurunawjan, 2022-09-22:
Hello,

I wanted to ask, I have NEO-D9C and ZED-F9P devices, how can I use the library for QZSS-L6 library? 

Is there anyone know how to use it? 

Thank you

PaulZC, 2022-09-22:
Hello NajwaKhamis (@lurunawjan ),

This is an interesting "challenge"! You are the first person we have spoken to who has a NEO-D9C. Congratulations!

The challenge is that: SparkFun is in the USA; I am in the UK; QZSS-L6 only covers Asia-Oceania; we do not have any NEO-D9C devices to test; and we do not have a QZSS-L6 simulator.

The library does support QZSS-L6 but it has not been tested.

**If** the NEO-D9C works the same as the NEO-D9S, then it should be possible to connect the NEO-D9C UART1 TX (TXD1) directly to the ZED-F9P UART2 RX. The correction data will be passed directly from the NEO to the ZED as UART serial data. But you will need to change the baud rate. The NEO defaults to 9600 baud and the ZED defaults to 38400 baud. To change the NEO to 38400 baud, you need to change the **CFG-UART1-BAUDRATE** configuration item to 38400. The easiest way to do this is to use [u-blox u-center](https://www.u-blox.com/en/product/u-center) and connect over USB.

You can also change the baud rate using this library. You can change it using I2C. The code would be similar to [Example30](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example30_NEO-D9S/Example30_NEO-D9S.ino). It would look like:

```
#include <SparkFun_u-blox_GNSS_Arduino_Library.h> //http://librarymanager/All#SparkFun_u-blox_GNSS
SFE_UBLOX_GNSS myQZSS; // NEO-D9C

void setup()
{
  Serial.begin(115200);
  Serial.println(F("NEO-D9C Example"));

  Wire.begin(); //Start I2C

  //=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
  // Begin and configure the NEO-D9C QZSS receiver

  //myQZSS.enableDebugging(); // Uncomment this line to enable helpful debug messages on Serial

  while (myQZSS.begin(Wire, 0x43) == false) //Connect to the u-blox NEO-D9C using Wire port. I think the D9C default I2C address is 0x43 (not 0x42) but I am not sure!
  {
    Serial.println(F("u-blox NEO-D9C not detected at default I2C address. Please check wiring."));
    delay(2000);
  }
  Serial.println(F("u-blox NEO-D9C connected"));

  myQZSS.setVal32(UBLOX_CFG_UART1_BAUDRATE, 38400); // match baudrate with ZED default
}

void loop()
{
}
```

It is also possible to read / receive the QZSS-L6 data from the NEO using I2C, and to send / push it to the ZED using I2C. To do that, you will need to use ```setRXMQZSSL6messageCallbackPtr```. The code will look similar to [Example19](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/ZED-F9P/Example19_LBand_Corrections_with_NEO-D9S) but will be much simpler. I do not think QZSS-L6 is encrypted, so you do not need to use keys.

I hope this lets you start working with the NEO-D9C. If you need more suggestions or ideas, please let me know.

Best wishes,
Paul

lurunawjan, 2022-09-23:
Thank you Sir Paul for your reply. 

I'm currently living in Japan and one of the company in Japan make the NEO-D9C device and I'm planning to use the NEO and ZED with a microcontroller. Does it mean if I connect the NEO and ZED using UART, then the connection for ZED and microcontroller is I2C and I'm using one of the Example to get the coordinate.  How can I identify that I'm using the QZSS-L6 data?

Thank you

PaulZC, 2022-09-23:
Hello NajwaKhamis (@lurunawjan ),

When the NEO receives QZSS-L6 correction data for the satellite, it will output a message (UBX-RXM-QZSSL6). That message can be used by the ZED for correction. The NEO message can be output on: I2C, UART1 or UART2. (The datasheet is not yet clear about UART2, but certainly the NEO-D9S can output messages on UART2.). For UART, it is possible to connect the NEO TX pin directly to the ZED RX pin. If the baud rates are the same, it will work. For I2C, you need a microcontroller to read the message from the NEO and send / push it to the ZED.

Can you please send me a link to the NEO and ZED boards you are using? That will help me give you more suggestions.

Best wishes,
Paul

PaulZC, 2022-09-23:
I found this web page:

https://www.geosense.co.jp/d9cx1_manual/

It contains some good information:

[image]

"The easiest and most efficient way is UART transfer, and the ZED-F9P module is wired as shown in the following diagram.
The actual power supply connects to the ZED-F9P module and powers the D9CX1 from that supply.
As the power supply terminal on the D9CX1 side, connect to the PWR terminal when the voltage on the ZED-F9P module side is 4V or less, and connect to the V5V terminal when it is 4V or more."

"Usually there is no problem with the above connection, but the u-blox NEO-D9C manual says to send position data from the ZED-F9P module to the NEO-D9C. This is to shorten the satellite acquisition time of the NEO-D9C, so if the NEO-D9C's satellite acquisition time is long, please also connect TX2 of the ZED-F9P module and RX2 of the D9CX1."

[image]

"Since this board has a Grove connector, it can be connected to a microcomputer equipped with a Grove connector such as ESPr without soldering.
Communication is done by I2C.
NEO-D9C's I2C address is 0x42 (factory default), which conflicts with ZED-F9P's address, so please change NEO-D9C's address to another address using u-center."

"When connected to ESP32, it has been confirmed that the following code will fix it properly. The NEO-D9C address is set to 0x41."

This is a surprise! I was expecting the NEO-D9C I2C address to be 0x43, not 0x42. 0x42 is the same as the ZED-F9P.

OK. I think I need to write a simple example for you - so we can check the I2C address.

PaulZC, 2022-09-23:
Hello NajwaKhamis (@lurunawjan ),

Please try this example:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/release_candidate/examples/Example32_NEO-D9C/Example32_NEO-D9C.ino

Please copy and paste some of the output here - I will be very interested to see it.

The example will show if you are receiving QZSS-L6 data and if it is error-free.

It also outputs the UBX-RXM-QZSSL6 message on both UART1 and UART2 at 38400 baud. You can connect NEO-D9C UART1 TX (TX1) to your ZED-F9P UART1 RX (RX1) or UART2 RX (RX2) and it will use the corrections. Also connect NEO GND to ZED GND.

Thank you,
Paul

lurunawjan, 2022-09-23:
Thank you Sir for the help.

[image]

the result as shown. should be the I2C from the micro-controller to the NEO right? I also change the wire port into 0x42

PaulZC, 2022-09-23:
Hello NajwaKhamis (@lurunawjan ),

Can you send me a photo of your boards? I would like to see how they are connected.

Can you send me a link for your NEO-D9C board? I would like to see the schematic if possible.

Please check you have only the NEO connected to I2C. If the NEO is 0x42, and you also have the ZED connected (also 0x42), there will be a 'collision'.

Thank you,
Paul

lurunawjan, 2022-09-23:
Hello Sir Paul,

I'm using teensy 4.0 as my micro-controller. I connect ZED to NEO by UART and from NEO to teensy via I2C. 

[image]
[image]
[d9cx1_schematics.pdf](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/files/9632855/d9cx1_schematics.pdf)

Thank you.

PaulZC, 2022-09-23:
Hello NajwaKhamis (@lurunawjan ),

Thank you for the photos. They are very clear.

Please try changing this line:

```
while (myQZSS.begin(Wire, 0x43) == false) //Connect to the u-blox NEO-D9C using Wire port. If 0x43 does not work, try 0x42
```

to:

```
while (myQZSS.begin(Wire, 0x43, 1100, true) == false) //Connect to the u-blox NEO-D9C using Wire port. If 0x43 does not work, try 0x42
```

It should let the NEO code "begin". Please try 0x43 first. Then try 0x42.

Thank you,
Paul

lurunawjan, 2022-09-23:
Hello Sir,

[image]

I got this result.

PaulZC, 2022-09-23:
It would be good to power the NEO board directly from the Teensy 5V pin.

Please try moving your red (5V) wire:

[image]

The brown wire (3.3V) is OK. Please do not move that.

Thank you.

lurunawjan, 2022-09-23:
Hello Sir,

[image]

I got this result. Does it means it got the correction data?

Thank you

PaulZC, 2022-09-23:
Hello NajwaKhamis (@lurunawjan ),

Yes! Congratulations! You have QZSS-L6 data!

Please tell me: which I2C address is working?

Thank you,
Paul

lurunawjan, 2022-09-23:
Hello Sir,

Thank you so much sir for your help.

It's 0x42 address. 

Can I measure the location of my coordinate with this connection? Like in example19

Thank you

PaulZC, 2022-09-23:
Hello NajwaKhamis (@lurunawjan ),

We have a small problem. The NEO is using I2C address 0x42. The ZED will also use address 0x42.

There are two solutions:

1) We can change the NEO to use a different I2C address (this is how the Geosense example works)
2) We can connect the ZED to the Teensy using Serial instead of I2C

I think solution (2) could be easier for you.

I will write a new example for you. It will take me a few hours. I think it will be ready for you tomorrow morning.

Have a good night,
Paul

PaulZC, 2022-09-23:
Hello NajwaKhamis (@lurunawjan ),

The new example is here:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/release_candidate/examples/ZED-F9P/Example22_QZSSL6_Corrections_with_NEO-D9C/Example22_QZSSL6_Corrections_with_NEO-D9C.ino

I have no way to test it, so I hope it works for you.

The connections are:

```
  Connections: e.g. for Teensy 4.0, Geosense D9CX1 NEO-D9C and SparkFun ZED-F9P:

  Teensy 5V (Vin) -> ZED-F9P 5V -> D9CX1 V5V (JP1 Pin 2)
  Teensy GND -> ZED-F9P GND -> D9CX1 GND (JP1 Pin 1)
  Teensy SDA1 (17) -> D9CX1 SDA (JP1 Pin 5)
  Teensy SCL1 (16) -> D9CX1 SCL (JP1 Pin 6)
  Teensy Serial1 TX1 (1) -> ZED-F9P UART1 RX1
  Teensy Serial1 RX1 (0) -> ZED-F9P UART1 TX1
  D9CX1 UART1 TX1 (JP1 Pin 3) -> ZED-F9P UART2 RX2
  D9CX1 UART1 RX1 (JP1 Pin 4) -> ZED-F9P UART2 TX2

  The Teensy communicates with the NEO-D9C (D9CX1) via I2C on address 0x42
  The Teensy communicates with the ZED-F9P via UART (Serial1 on Teensy, UART1 on ZED) to avoid the I2C address collision
  The NEO-D9C corrections (UBX-RXM-QZSSL6) are sent from NEO UART1 to ZED UART2
```

Please try it and copy-and-paste the results here.

Good luck!
Paul

lurunawjan, 2022-09-24:
Hello Sir Paul,

[image]

I got this result. 

Thank you

PaulZC, 2022-09-24:
Hello NajwaKhamis (@lurunawjan ),

It looks like both NEO-D9C and ZED-F9P are communicating well. But the NEO is not receiving a signal and sending any correction data messages.

Please leave the example running. Do you see the "New QZSS-L6 data received" messages, like yesterday?

Thank you,
Paul

lurunawjan, 2022-09-24:
Hello Sir,

Yes I saw the "New QZSS-L6 data received" like yesterday. Maybe because I put the antenna not in very good sky view. It will affect right?

Thank you

PaulZC, 2022-09-24:
Yes, the antenna needs a clear view of the sky.

Please leave the example running for a few minutes and then copy and paste the messages here.

Thank you. (This is fun!)

lurunawjan, 2022-09-28:
Hello Sir,

Sorry for the late update. I put the antenna on the roof of the car for a few minutes and I got this result. It didn't detect my location. 

Thank you

[image]

lurunawjan, 2022-09-28:
Also this result
[image]

lurunawjan, 2022-09-28:
This is the final result after a few minutes 

[image]

PaulZC, 2022-09-28:
Hello NajwaKhamis (@lurunawjan ),

Everything is working almost perfectly.

The NEO-D9C is receiving a signal with good carrier/noise (>30dB), on both channels (A + B), from multiple satellites (4 + 7), with zero bit errors.

The ZED is using the corrections: UBX-RXM-COR protocol is QZSSL6, it is error-free and is "used".

You have a "Fixed" carrier solution with a horizontal accuracy of 35mm (absolute). That is excellent.

The only surprise is that the channel B messages from satellite 7 are "erroneous". I do not know why. I will need to ask a friend for advice.

Congratulations!

Very best wishes,
Paul

mazgch, 2022-09-28:
What antenna are you using? CLAS is on L6 it is really important that you have a good antenna and view to sky. I have seen much better CNO values while in Japan or testing on a similator. Low 30s might be at the data decoding limit.

lurunawjan, 2022-09-28:
@PaulZC 

Thank you so much Sir for your help.

@mazgch 
Hello Sir, I'm using L6 band antenna with the good sky view. 

Thank you.

PaulZC, 2022-09-29:
Hello NajwaKhamis (@lurunawjan ),

Can you please tell us the make (manufacturer) and model of your antenna?

Can you also close jumper JP4 on the NEO-D9C board. That will help provide better 3.3V to the antenna. (With jumper JP4 open, only the ZED-F9P is providing power. With JP4 closed, both boards will provide power. The antenna may need more current than the ZED-F9P board can provide.)

Thank you,
Paul

lurunawjan, 2022-10-11:
Hello Sir,

The antenna model that I use is antenna BT-345AJ.

PaulZC, 2022-10-17:
Hello NajwaKhamis (@lurunawjan ),

The two new examples are included in version 2.2.17 - released today.

I am going to close this issue, but please reopen it if you need more help.

Best wishes,
Paul

closed 2022-10-17 by PaulZC