sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #159: Using Hardware UART

jan5650, 2022-09-23:
Hi Guys!

Using hardware UART is possible?
I am using Arduino Mega 2560, wich got 3 hardware UART. How can i define and use hardware UART like Serial1?

```
#include <SparkFun_u-blox_GNSS_Arduino_Library.h>
SFE_UBLOX_GNSS myGNSS;
#define mySerial(Serial1);

void setup() {
do {
    mySerial.begin(38400);
    if (myGNSS.begin(mySerial) == true) break;

    delay(100);
    mySerial.begin(9600);
    if (myGNSS.begin(mySerial) == true) {
        myGNSS.setSerialRate(38400);
        delay(100);
    } else {
        myGNSS.factoryReset();
        delay(2000); 
    }
  } while(1);
 }
```
 
It won't work, "'mySerial' was not declared in this scope" error got

Can you help me?
Thank you!

PaulZC, 2022-09-23:
Hi @jan5650 ,

Change:

```
#define mySerial(Serial1);
```

to:

```
#define mySerial Serial1
```

and you should be good to go.

Best wishes,
Paul

closed 2022-09-23 by PaulZC

jan5650, 2022-09-24:
Sadly it's not working:

```
#include <SparkFun_u-blox_GNSS_Arduino_Library.h>
SFE_UBLOX_GNSS myGNSS;
#define mySerial(Serial1);

void setup() {
do {
    mySerial.begin(38400);
    if (myGNSS.begin(mySerial) == true) break;

    delay(100);
    mySerial.begin(9600);
    if (myGNSS.begin(mySerial) == true) {
        myGNSS.setSerialRate(38400);
        delay(100);
    } else {
        myGNSS.factoryReset();
        delay(2000); 
    }
  } while(1);
 }
```

**exit status 1 expected primary-expression before '.' token** in the "mySerial.begin(38400);" line

PaulZC, 2022-09-24:
You have a typo. Please remove the semicolon:

```
#define mySerial Serial1
```

jan5650, 2022-09-24:
Great, thank you!