sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #160: Help wanted: Cant get my neo-m8p to print out rtcm correction data

JacobWilkins, 2022-10-12:
I've been using the example 2 for the neo-m8p called Example2_StartRTCMBase. The example works fine except for the last function that is supposed to then print the RTCM data, it doesn't get called and I’m unable to figure out why. 

 I’m using the neo-m8p along with a sparkfun blackboard and I’m using an I2c cable to connect them.

I’m using a mac and a usb-c cable to power the blackboard and am using the arduino editor.

To reproduce this issue just open Example2_StartRTCMBase and run it with a neo-m8p and if its working the same as mine then the final function won't be getting called and wont print the RTCM data in the serial monitor as it is supposed to.

[image]

PaulZC, 2022-10-12:
Hi @JacobWilkins ,

I don't have a NEO-M8P here to test... But looking at the example code, I _think_ there's a bug in there...

Please try changing Line 49:

From:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/6f98c6c27a271e130ede85f572577186cac6962f/examples/NEO-M8P-2/Example2_StartRTCMBase/Example2_StartRTCMBase.ino#L49

To:

```
myGNSS.setI2COutput( COM_TYPE_UBX | COM_TYPE_NMEA | COM_TYPE_RTCM3 ); //Set the I2C port to output UBX, NMEA and RTCM3
```

Do you see RTCM now?

Best wishes,
Paul

JacobWilkins, 2022-10-14:
Hi @PaulZC,

I changed line 49 and am now seeing the RTCM data, thank you for your help.

Jake

PaulZC, 2022-10-14:
Hi Jake (@JacobWilkins ),

Thanks for the update - I'm glad that's working for you!

Just re-opening this. It will be my reminder to update the examples!

Best wishes,
Paul

PaulZC, 2022-10-17:
Hi Jake (@JacobWilkins ),

This has been resolved in release v2.2.17. There were five examples that needed updating...! Thanks again for the heads-up.

Best wishes,
Paul

closed 2022-10-17 by PaulZC