sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #171: UBX-NAV-SIG

baldvinhansson, 2022-12-02:
Thank you all for your continued excellent work on this library! I am looking for dBHz data available via UBX-NAV-SIG. Am I correct in finding that this library does not support that message at present time? If so, are there any plans to add it? Would adding it not be practical for some reason?

PaulZC, 2022-12-04:
Hi @baldvinhansson ,

Correct - the library does not fully support UBX-NAV-SIG at the present time. Adding it is not _currently_ practical due to memory constraints. Please see this pull request for more details: #169 

However, you can use a custom command to manually request the SIG data and parse it yourself. Please see examples 20 and 21 for more details. Issue #52 contains helpful information too.

We are working on a new version of the library where it will be _much_ easier to add new messages, and delete messages too if required to save memory.

Best wishes,
Paul

PaulZC, 2022-12-04:
Closing... Please re-open if you need more help or advice.

closed 2022-12-04 by PaulZC

PaulZC, 2022-12-18:
Hi @baldvinhansson ,

Just to follow up on this:

**cno** (dBHz) is available in the UBX_NAV_SAT message, and the library does provide full support for that.

Please have a look at [CallbackExample7](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Callbacks/CallbackExample7_NAV_SAT/CallbackExample7_NAV_SAT.ino).

Best wishes,
Paul

baldvinhansson, 2022-12-18:
Thank you for this hint! And I'm happy to tell you that I followed your earlier advice and managed to implement what I needed as a custom function. I'm looking forward to continuing using the library and its future versions in my projects.