sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #173: Baudrate

jan5650, 2022-12-16:
Hello!

I would like to ask you if a baud rate of 38400 would be enough for the following data volume?
_5 Hz sampling from GPS, Galileo,Glonass_

```
  myGNSS.setMeasurementRate(200); 
  myGNSS.setNavigationRate(1); 
  myGNSS.setAutoPVTrate(1); 

  myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_GPS); 
  myGNSS.enableGNSS(false, SFE_UBLOX_GNSS_ID_BEIDOU); 
  myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_GALILEO); 
  myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_GLONASS); 
  
  myGNSS.setUART1Output(COM_TYPE_UBX); 
  myGNSS.setI2COutput(COM_TYPE_UBX); 
  myGNSS.saveConfiguration();   
```
Another question:

`myGNSS.saveConfiguration();   `

For Arduino, the EEPROM can be overwritten a limited number of times. If this is reached, the EEPROM will be corrupted.

The above function is used to write to the GPS modules's EEPROM. How many times can we do this in order of magnitude? 
I guess it can handle 1000-2000 writes at least, isn't?

PaulZC, 2022-12-16:
Hi @jan5650 

1) It depends on how many messages you have enabled. 38400 baud is _approximately_ 3800 bytes or characters per second. So, at 5Hz, you're limited to ~760 characters. With all of the standard NMEA messages enabled for all constellations, you will probably exceed that.

2) It depends if the data is being stored in flash or battery-backed RAM. The module datasheet should include this information?

Best wishes,
Paul

closed 2022-12-16 by PaulZC

PaulZC, 2022-12-16:
If you have disabled all NMEA messages… The UBX NAV PVT message is 100 bytes. If that’s the only message you have enabled, then you should be ok.

jan5650, 2023-02-10:
> If you have disabled all NMEA messages… The UBX NAV PVT message is 100 bytes. If that’s the only message you have enabled, then you should be ok.

Unfortunately, I'm not really into this GPS stuff.
This is in the setup section:

```

  myGNSS.setMeasurementRate(200); 
  myGNSS.setNavigationRate(1); 
  myGNSS.setAutoPVTrate(1); 

  myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_GPS); 
  myGNSS.enableGNSS(false, SFE_UBLOX_GNSS_ID_BEIDOU); 
  myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_GALILEO); 
  myGNSS.enableGNSS(true, SFE_UBLOX_GNSS_ID_GLONASS); 
  
  myGNSS.setUART1Output(COM_TYPE_UBX); 
  myGNSS.setI2COutput(COM_TYPE_UBX); 
  myGNSS.saveConfiguration(); 
```
So i turn only on GPS,GLONASS?GALIELO
I turn off NMEA sentences, only UBX used
I set myGNSS.setAutoPVTrate(1); 

**- What exatly myGNSS.setAutoPVTrate(1); do?**

In the loop i used only :

```
 myGNSS.checkUblox();
  satelites = myGNSS.getSIV();
  lan = myGNSS.getLatitude();
  lon = myGNSS.getLongitude();
```

**- How myGNSS.checkUblox(); and myGNSS.setAutoPVTrate(1);  related to each other?**

In this case (no nmea's are coming, only UBX) baud rate 38400 would be enough?
Maybe myGNSS.checkUblox(); check that there is new data, then get the UBX, and from that get LATITUDE and LONGTITUDE?

PaulZC, 2023-02-11:
Hello @jan5650 ,

> Unfortunately, I'm not really into this GPS stuff.

Don't worry. We all have to start somewhere! Just for information, we use GitHub Issues to report and correct true Issues (bugs) in the code. It is not really the right place to ask for help. The [GPS Forum](https://forum.sparkfun.com/viewforum.php?f=116) is the best place for that.

> What exatly myGNSS.setAutoPVTrate(1); do?

It tells the module to output the PVT message automatically every 1 navigation cycles. If you set it to 2, it will output PVT every second (2nd) cycle. 3, every third (3rd) cycle.

> How myGNSS.checkUblox(); and myGNSS.setAutoPVTrate(1); related to each other?

They are different. ```checkUblox``` checks for the arrival of new data from the module. In your case, it checks to see if a new PVT message has arrived and will process it if it has. ```getSIV``` will then return the number of satellites in the navigation solution, extracted from the new PVT message. If a new message has not arrived, ```getSIV``` will return old (stale) data from the previous PVT message.

If you want to make sure you are reading new (fresh) data each time, then call ```getPVT()``` and wait for it to return ```true``` :

```
if (myGNSS.getPVT()) // getPVT returns true if fresh data is available
{
  satelites = myGNSS.getSIV();
  lan = myGNSS.getLatitude();
  lon = myGNSS.getLongitude();
}
```

It's confusing, but calling ```getPVT``` will call ```checkUblox``` too. You don't then need to call them separately.

Anyway, to answer your original question:

The module is outputting the PVT message only.
PVT is 100 bytes long. 100 bytes is 1000 bits. (1 start bit, 8 data bits, 1 stop bit)
At 5Hz, the module is outputting 5000 bits per second.
This is less than 38400.
So 38400 baud is OK.

Best wishes,
Paul