sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #178: SparkFun GNSS Receiver Breakout  - MAX-M10S issues with your GPS library

roblatour, 2023-01-17:
I bought a SparkFun GNSS Receiver Breakout - MAX-M10S (Qwiic), and am connecting to an ESP32 board (OLIMEX ESP32 ISO POE) via the RX, TX, and PPS pins (not using IC2).

Originally, I had a sketch which I had written using the tinyGPSPlus library.  With it I could connect to the board at 9600 BAUD and get the data (just time and date) I needed from the GPS no problem. 

While this was working well, there was something I wanted to improve that it was not evident to me how to do with the tinyGPSPlus library, so based on the link on your sales page for the module I bought ( https://www.sparkfun.com/products/18037 ) I tried the library and example here: 
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/Example15_GetDateTime
to see if I could use it as a replacement.

That is when the problems began.

While your library and the Example15_GetDateTime example worked ok, when I tried to revert back to using my original sketch it would no longer work (although it has remained unchanged) with the GPS.

I noticed the SparkFun GetDateTime example, as part of the example, saved the configuration to the board using a baud rate of 34800.   However, even with tweaking, I could not get the SparkFun GetDateTime example to revert its baud rate to 9600.

I tried setting the baud rate in my tinyGPSPlus library to 38400 but it does not retrieve data from the Sparkfun board at that speed either.

So in short, after having your example code from your library library my board is no longer usable with the tinyGPSPlus library or my sketch. 

I also tried running your  https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/Example11_ResetModule/Example2_FactoryDefaultsviaSerial
example, tweaking it to only run case 2 one time; but it doesn't work at all - rather just crashes the processor.

Please advise how can I reset my SparkFun GNSS Receiver Breakout - MAX-M10S (Qwiic) to its original factory out of the box settings such that I can use it with my original sketch once again.

PaulZC, 2023-01-17:
Hello Rob (@roblatour ),

The u-blox modules support two protocols: NMEA and UBX. NMEA is human-readable ASCII text. UBX is a much more sophisticated binary protocol (not human readable). tinyGPSPlus uses NMEA; it parses the latitude, longitude etc. from the NMEA text. Most of the examples in our library use UBX - and actually disable NMEA because it's essentially redundant.

Example2_FactoryDefaultsviaSerial unfortunately does the same thing. There's a line in the code which turns off the NMEA:

```
myGNSS.setUART1Output(COM_TYPE_UBX); //Set the UART port to output UBX only
```

OK. The following code will reset the module and get you going again. Please copy and paste it into a blank sketch and upload to your ESP board. I'm assuming the OLIMEX board uses ```Serial1``` for its U1TXD and U1RXD pins? If not, let me know and I'll work out different code for you.

```
#include <SparkFun_u-blox_GNSS_Arduino_Library.h> //http://librarymanager/All#SparkFun_u-blox_GNSS
SFE_UBLOX_GNSS myGNSS;

#define gnssSerial Serial1 // Use Serial1 to connect to u-blox. Change if required
#define gnssBaud 38400 // Use 38400 baud to connect to u-blox. Change if required

void setup()
{
  delay(1000); // Wait for 1 second before starting Serial

  Serial.begin(115200); // Start Serial (USB) at 115200 baud

  gnssSerial.begin(gnssBaud); // Start serial comms for the u-blox module

  while (myGNSS.begin(gnssSerial) == false) // Begin communication with the module
  {
    Serial.println("Could not begin the u-blox module. Please check gnssSerial and gnssBaud.");
  }

  Serial.println("u-blox module detected. Resetting...");

  myGNSS.factoryDefault(); // Reset everything to the factory defaults. Save the settings in battery-backed RAM

  delay(5000); // Wait for the module to restart

  Serial.println("Reset complete. Enjoy your NMEA!");
}

void loop()
{
  // Nothing to do here
}
```

I believe the MAX-M10S defaults to 38400 baud, certainly the one I have here does. So, after the reset, I believe you will need to use 38400 for tinyGPSPlus.

I hope this gets you going again. Let me know if you need more help.

Best wishes,
Paul

roblatour, 2023-01-17:
Thank you for your quick reply.

I ran the sketch as above, and it displayed the messages "u-blox module detected. Resetting..." and "Reset complete. Enjoy your NMEA!".

After that, I reloaded my original sketch worked as it had before (with the baud rate on the device at 9600 - I initially tried 34800 but it didn't work).

 In any case, thank you for your support, you are welcome to close this ticket.

However, may I suggest, either putting some comments in the example sketches in the library about what happens in this regard when you use them and/or also including the sketch you provided above in with the other examples.

Again, with thanks.

PaulZC, 2023-01-17:
Thanks for the update Rob - glad that's working for you.

All the best,
Paul

closed 2023-01-17 by PaulZC