sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #191: Arduino Due slower update rate than Uno over I2C

smithan7, 2023-04-17:
Thanks for a great library and please let me know if there are any details I am leaving out. I searched for old issues on this and the closest I found was Arduino DUE Setup #66 - which was related to Serial connections.

### Subject of the issue
I am using the Arduino Due with the SparkFun GPS Dead Reckoning Breakout - NEO-M8U (Qwiic) with the unmodified Example3_GetPosition.

I am wired to the Due using pins 20, 21, 3.3v, and GND (without pullup resistors) and I am able to connect and get GPS data to print on the screen as shown below:

********************* Begin Serial Monitor from Due **********************************************
05:46:41.205 -> SparkFun u-blox Example
05:47:05.403 -> Lat: 394265215 Long: -1198031269 (degrees * 10^-7) Alt: 1470719 (mm) SIV: 12
05:47:11.551 -> Lat: 394265207 Long: -1198031285 (degrees * 10^-7) Alt: 1470172 (mm) SIV: 12
05:47:17.535 -> Lat: 394265230 Long: -1198031291 (degrees * 10^-7) Alt: 1470268 (mm) SIV: 12
05:47:23.936 -> Lat: 394265240 Long: -1198031258 (degrees * 10^-7) Alt: 1470107 (mm) SIV: 12
05:47:30.635 -> Lat: 394265252 Long: -1198031239 (degrees * 10^-7) Alt: 1469609 (mm) SIV: 13
05:47:37.268 -> Lat: 394265252 Long: -1198031236 (degrees * 10^-7) Alt: 1468796 (mm) SIV: 13
05:47:43.851 -> Lat: 394265249 Long: -1198031244 (degrees * 10^-7) Alt: 1468349 (mm) SIV: 13
05:47:50.466 -> Lat: 394265249 Long: -1198031239 (degrees * 10^-7) Alt: 1467867 (mm) SIV: 14
05:47:57.183 -> Lat: 394265235 Long: -1198031253 (degrees * 10^-7) Alt: 1467201 (mm) SIV: 14
05:48:03.682 -> Lat: 394265197 Long: -1198031219 (degrees * 10^-7) Alt: 1466472 (mm) SIV: 14
05:48:10.215 -> Lat: 394265186 Long: -1198031161 (degrees * 10^-7) Alt: 1465530 (mm) SIV: 14
05:48:10.332 -> Lat: 394265187 Long: -1198031112 (degrees * 10^-7) Alt: 1464987 (mm) SIV: 14
05:48:17.731 -> Lat: 394265179 Long: -1198031109 (degrees * 10^-7) Alt: 1464991 (mm) SIV: 14
05:48:24.296 -> Lat: 394265165 Long: -1198031100 (degrees * 10^-7) Alt: 1464528 (mm) SIV: 14
05:48:30.797 -> Lat: 394265151 Long: -1198031117 (degrees * 10^-7) Alt: 1463669 (mm) SIV: 14
05:48:37.113 -> Lat: 394265120 Long: -1198031108 (degrees * 10^-7) Alt: 1462705 (mm) SIV: 14
05:48:43.379 -> Lat: 394265088 Long: -1198031115 (degrees * 10^-7) Alt: 1462157 (mm) SIV: 14
05:48:43.596 -> Lat: 394265044 Long: -1198031095 (degrees * 10^-7) Alt: 1461412 (mm) SIV: 14
05:48:50.645 -> Lat: 394265036 Long: -1198031094 (degrees * 10^-7) Alt: 1461171 (mm) SIV: 14
05:48:56.912 -> Lat: 394264992 Long: -1198031098 (degrees * 10^-7) Alt: 1459645 (mm) SIV: 14 
********************* End Serial Monitor from Due **********************************************

The issue is that it runs really slowly, from the timestamps it takes ~6s for each update. I originally had this on a Sparkfun redboard Qwiic (~Arduino Uno) (hooked up A5, A4, 3.3v, and GND without pullup resistors) and re-running using the exact same Example3_GetPosition I get:

********************* Begin Serial Monitor from Uno **********************************************
05:43:30.968 -> SparkFun u-blox Example
05:43:32.088 -> Lat: 394265918 Long: -1198031010 (degrees * 10^-7) Alt: 1471660 (mm) SIV: 0
05:43:33.087 -> Lat: 394265899 Long: -1198031022 (degrees * 10^-7) Alt: 1471503 (mm) SIV: 0
05:43:34.087 -> Lat: 394265880 Long: -1198031034 (degrees * 10^-7) Alt: 1471345 (mm) SIV: 0
05:43:35.087 -> Lat: 394265860 Long: -1198031045 (degrees * 10^-7) Alt: 1471188 (mm) SIV: 0
05:43:36.070 -> Lat: 394265841 Long: -1198031057 (degrees * 10^-7) Alt: 1471030 (mm) SIV: 0
05:43:37.103 -> Lat: 394265821 Long: -1198031069 (degrees * 10^-7) Alt: 1470873 (mm) SIV: 0
05:43:38.087 -> Lat: 394265802 Long: -1198031081 (degrees * 10^-7) Alt: 1470715 (mm) SIV: 0
05:43:39.087 -> Lat: 394265783 Long: -1198031093 (degrees * 10^-7) Alt: 1470558 (mm) SIV: 0
05:43:40.104 -> Lat: 394265763 Long: -1198031105 (degrees * 10^-7) Alt: 1470400 (mm) SIV: 0
05:43:41.103 -> Lat: 394265756 Long: -1198031109 (degrees * 10^-7) Alt: 1470337 (mm) SIV: 0
05:43:42.103 -> Lat: 394265756 Long: -1198031109 (degrees * 10^-7) Alt: 1470337 (mm) SIV: 0
05:43:43.086 -> Lat: 394265756 Long: -1198031109 (degrees * 10^-7) Alt: 1470337 (mm) SIV: 0
05:43:44.084 -> Lat: 394265756 Long: -1198031109 (degrees * 10^-7) Alt: 1470337 (mm) SIV: 0
05:43:45.103 -> Lat: 394265756 Long: -1198031109 (degrees * 10^-7) Alt: 1470337 (mm) SIV: 0
05:43:46.120 -> Lat: 394265756 Long: -1198031109 (degrees * 10^-7) Alt: 1470337 (mm) SIV: 0 
********************* End Serial Monitor from Uno **********************************************

This is a simple example, I noticed the slowness on a separate script and am using this to demonstrate a re-creatable difference on a stock example. I have tried increasing the wire speed to 400,000 Hz (Wire.setClock(400000);) and that didn't help. I also notice that the Due is less reliable in detecting the GPS receiver, which I assume is a timeout issue. I have also used the examples to modify the GPS update behavior and on the Uno it behaves as expected and updates but on the Due it frequently does not update OR when it does it does not effect the actual update rate. 

I tested running print statements on the Due and it does not lag - it appears to be something in either the i2c or GPS libraries/connections inducing the lag.

### Your workbench
* What development board or microcontroller are you using?
          * I am using a Sparkfun Redboard Qwiic (~Uno) and Arduino Due
* What version of hardware or breakout board are you using? 
         * SparkFun GPS Dead Reckoning Breakout - NEO-M8U (Qwiic)
* How is the breakout board wired to your microcontroller?
         * Redboard UNO - A4, A5, 3.3v, GND pins without pullup resistors - direct from Arduino female headers to Qwiic  connector
              * Have also used the built-in Qwiic connector on the Redboard Uno direct to the GPS Qwiic without issue and went back to this to test the cable used to connect the Due. 
         * Arduino Due - 20, 21, 3.3v, GND pins without pullup resistors - direct from Arduino female headers to Qwiic  connector
* How is everything being powered?
         * Both boards are powered via USB port to micro-USB connector from my laptop to the board.
* Are there any additional details that may help us help you?
         * Not that I can think of... 

[... 82 lines cut ...]

    long altitude = myGNSS.getAltitude();
    Serial.print(F(" Alt: "));
    Serial.print(altitude);
    Serial.print(F(" (mm)"));

    byte SIV = myGNSS.getSIV();
    Serial.print(F(" SIV: "));
    Serial.print(SIV);

    Serial.println();
  }
}

***************************************************************************************************************
***************************************************************************************************************
*************************************End Code ***************************************************************
***************************************************************************************************************
***************************************************************************************************************

### Expected behavior
I expected the Arduino Due to have the same update rate as the Uno.

### Actual behavior
The Due is significantly slower than the Uno.

PaulZC, 2023-04-17:
Hi @smithan7 ,

I think you are seeing the same thing as this issue: https://forum.sparkfun.com/viewtopic.php?p=239338#p239338 . The u-blox modules use I2C clock stretching, and the Due does not support clock stretching...

Switching to Serial or SPI should solve this issue for you.

I'm going to close this, but please re-open if you need more help or suggestions.

Best wishes,
Paul

closed 2023-04-17 by PaulZC

smithan7, 2023-04-19:
Thanks for giving me the answer - I am now trying to use the Serial1 port of the Due to make the connection but have not had any luck getting it to connect. I used the same program from https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/66

But couldn't get a connection. I looked at the U-Blox Neo M8U protocol sheet and it listed the following possible baud rates:
[image]

I stepped through and tried each one without getting past:

*************
SparkFun u-blox Example
..............................................................
*************

I checked again over i2c and there I am able to pull pvt data from the module.

My connections are from the Arduino Due Tx1 (18) -> GPS Rx, Due Rx1(19) -> GPS Tx, Due 3.3v -> GPS 3.3v, Due GND -> GPS GNG.

here is the complete script I am using and change out the baud rate for Serial1:

#include <Wire.h> //Needed for I2C to GNSS
#include <SparkFun_u-blox_GNSS_Arduino_Library.h>
SFE_UBLOX_GNSS myGNSS;

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println("SparkFun u-blox Example");
  Serial1.begin(460800);
  myGNSS.begin(Serial1);
  myGNSS.assumeAutoPVT(true, true);

}

void loop()
{
  if (myGNSS.getPVT())
  {
    Serial.println();

    long latitude = myGNSS.getLatitude();
    Serial.print(F("Lat: "));
    Serial.print(latitude);

    long longitude = myGNSS.getLongitude();
    Serial.print(F(" Long: "));
    Serial.print(longitude);
    Serial.print(F(" (degrees * 10^-7)"));

    long altitude = myGNSS.getAltitude();
    Serial.print(F(" Alt: "));
    Serial.print(altitude);
    Serial.print(F(" (mm)"));

    byte SIV = myGNSS.getSIV();
    Serial.print(F(" SIV: "));
    Serial.print(SIV);

    Serial.println();
  }
  else
  {
    delay(5);
    static int counter = 0;
    counter++;
    if (counter > 10)
    {
      Serial.print(".");
      counter = 0;
    }
  }
}

Is there something I am missing to enable the Serial on the GPS module? I haven't seen anything but am not sure what else to try. The common guidance is to make sure I am using the right baud rate and get a clear view of the sky. I have verified it gets PVT data river i2c so I would assume I have a clear view of the sky and I have tried every baud rate in the protocol description. Anything I am missing? 

To me, this didn't seem worthy of opening a new issue but I will if that is preferred. Thanks again.

PaulZC, 2023-04-19:
Hi @smithan7 ,

I don't have a Due with me, so all I can offer are suggestions which you can test.

The NEO-M8U defaults to 9600 Baud. 9600 should work perfectly. You should not need to change it.

This example is a better starting point:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example13_PVT/Example3_AutoPVTviaUart/Example3_AutoPVTviaUart.ino

You are using hardware Serial port Serial1 (pins 18 and 19), so the definition of ```mySerial``` becomes:

```
#define mySerial Serial1
```

Set the baud rate using:

```
#define baudRate 9600
```

I think the main problem is that you are using ```myGNSS.assumeAutoPVT(true, true);```. Please see [this comment](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/66#issuecomment-917481563).

Best wishes,
Paul

smithan7, 2023-04-19:
Hi Paul,

Thanks for your assistance on this. I made the changes you recommended and used:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/examples/Example13_PVT/Example3_AutoPVTviaUart/Example3_AutoPVTviaUart.ino

with the following changes to use Serial1:

*****************************************
//#include <SoftwareSerial.h>

#define mySerial Serial1 // Uncomment this line to connect via Serial2
// - or -
//SoftwareSerial mySerial(10, 11); // Uncomment this line to connect via SoftwareSerial(RX, TX). Connect pin 10 to GNSS TX pin.

#define baudRate 9600 // Uncomment this line to select 9600 Baud for the M8
// - or -
//#define baudRate 38400 // Uncomment this line to select 38400 Baud for the F9

void setup()
{
  Serial.begin(115200);
  while (!Serial); //Wait for user to open terminal
  Serial.println("SparkFun u-blox Example");

  mySerial.begin(baudRate); // Start the Serial port

  if (myGNSS.begin(mySerial) == false) //Connect to the u-blox module using Serial
  {
    Serial.println(F("u-blox GNSS not detected. Please check wiring. Freezing."));
    while (1);
  }

  myGNSS.setUART1Output(COM_TYPE_UBX); //Set the UART1 port to output UBX only (turn off NMEA noise)
  myGNSS.setNavigationFrequency(2); //Produce two solutions per second
  myGNSS.setAutoPVT(true); //Tell the GNSS to "send" each solution
  //myGNSS.saveConfiguration(); //Optional: Save the current settings to flash and BBR
}
****************************************

It is not detecting the GPS and is hanging at:

   SparkFun u-blox Example
   u-blox GNSS not detected. Please check wiring. Freezing.

I tried powering it direct to the 3.3v and gnd pins adjacent to the Rx/Tx on the board and through the i2c port (both acquired and got the blue led for PPS (I assume they're connected on the board but am running out of ideas). I also tried using Serial2 on the Due which had the same result. I verified my jumper cables are ok by using them instead of the Qwiic port on the GPS and going direct to the SDA/SCL pins on the GPS - which worked using the same jumper cables that are not working with the Serial connection.

By my understanding, this should just work at this point and I am not sure what else to try. Really appreciate your patience, quick responses, and help on this. If you see anything I am missing or anything else I might test please let me know.

smithan7, 2023-04-20:
That worked!!!!

Thanks for all of your help Paul, I genuinely appreciate your patience and support on this.

SparkFun u-blox Example
....
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
.......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
.......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
.......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
.......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
...........
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
.......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
............
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
......
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0
.............
Lat: 115618923 Long: -1386011952 (degrees * 10^-7) Alt: 852144 (mm) SIV: 0
.....
Lat: 115617280 Long: -1386003589 (degrees * 10^-7) Alt: 852120 (mm) SIV: 0
...........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
....
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.............
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
..........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
.........
Lat: 394272490 Long: -1198014747 (degrees * 10^-7) Alt: 1314406 (mm) SIV: 0
..........
Lat: 394265927 Long: -1198030831 (degrees * 10^-7) Alt: 1455876 (mm) SIV: 0
.........
Lat: 394265934 Long: -1198030831 (degrees * 10^-7) Alt: 1455838 (mm) SIV: 0
.........
Lat: 394265913 Long: -1198030834 (degrees * 10^-7) Alt: 1456119 (mm) SIV: 0
.........
Lat: 394265905 Long: -1198030820 (degrees * 10^-7) Alt: 1456724 (mm) SIV: 0
.........
Lat: 394265880 Long: -1198030837 (degrees * 10^-7) Alt: 1457195 (mm) SIV: 0
.........
Lat: 394265918 Long: -1198030796 (degrees * 10^-7) Alt: 1457046 (mm) SIV: 0

PaulZC, 2023-04-20:
Hi @smithan7 ,

No problem. Glad that's working for you!

Best wishes,
Paul