sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #196: NMEA packets are output at different rates when interfaced over I2C with Sam-M8Q

mpvlad, 2023-05-11:
### Subject of the issue
NMEA packets obtained from SAM-M8Q over I2C interface 

### Your workbench
* Mega, level shifter and spark fun SAM-M8Q breakout board 
* Connections are done via jumpers 
* GPS module powered from 3.3V of mega board 

### Steps to reproduce
Used Example1_BasicNMEARead.ino as the starting point 

Add the following to the setup function

````
myGPS.disableNMEAMessage(UBX_NMEA_GLL, COM_PORT_I2C);
  myGPS.disableNMEAMessage(UBX_NMEA_VTG, COM_PORT_I2C);
  myGPS.disableNMEAMessage(UBX_NMEA_TXT, COM_PORT_I2C);
  myGPS.disableNMEAMessage(UBX_NMEA_GLL, COM_PORT_I2C);
  myGPS.disableNMEAMessage(UBX_NMEA_GSA, COM_PORT_I2C);
  myGPS.disableNMEAMessage(UBX_NMEA_GSV, COM_PORT_I2C);

  myGPS.setI2COutput(COM_TYPE_NMEA);
  myGPS.configureMessage(UBX_CLASS_NAV, UBX_NAV_PVT, COM_PORT_I2C, 0);

  myGPS.setNavigationFrequency(10); //Set output to 10 times a second

  myGPS.saveConfiguration();

  //This will pipe all NMEA sentences to the serial port so we can see them
  myGPS.setNMEAOutputPort(Serial);
````

### Expected behavior
RMC and GGA packets are printed out at the same rate, in this case 10Hz 

### Actual behavior
RMC packet is provided at ~10Hz as expected 
GGA packet outputs at random rates, sometimes it looks as if it is outputting at less than 1Hz

PaulZC, 2023-05-11:
Hi @mpvlad ,

Try deleting or reducing the delay(250) in the loop.

Best wishes,
Paul

closed 2023-05-11 by PaulZC

mpvlad, 2023-05-11:
@PaulZC I've had the delay set below 100ms, forgot to mention that in the original issue. Like I said, RMC is printing out at the expected 10Hz however GGA is random. Sometimes for a few seconds they are provided at the same rate sometimes it takes seconds before I see another GGA message printed out. RMC is not affected.

PaulZC, 2023-05-11:
Hi @mpvlad ,

If you still have UART1 enabled - and still have the default messages enabled on UART1 - you will be overloading it. It can have unexpected side-effects.

Please try disabling UART1: ```myGNSS.setUART1Output(0);```

I hope this helps,
Paul

PaulZC, 2023-05-11:
Yes, that was it.

If you see any ```$GNTXT,01,01,00,txbuf alloc*61``` messages, that's a clue that the module is not happy and that you are overloading one or more ports. If you disable UART1, you should see:

```
22:41:44.647 -> $GNRMC,214145.60,V,,,,,,,110523,,,N*66
22:41:44.647 -> $GNGGA,214145.60,,,,,0,00,99.99,,,,,,*79
22:41:44.732 -> $GNRMC,214145.70,V,,,,,,,110523,,,N*67
22:41:44.732 -> $GNGGA,214145.70,,,,,0,00,99.99,,,,,,*78
22:41:44.874 -> $GNRMC,214145.80,V,,,,,,,110523,,,N*68
22:41:44.874 -> $GNGGA,214145.80,,,,,0,00,99.99,,,,,,*77
22:41:44.980 -> $GNRMC,214145.90,V,,,,,,,110523,,,N*69
22:41:44.980 -> $GNGGA,214145.90,,,,,0,00,99.99,,,,,,*76
22:41:45.051 -> $GNRMC,214146.00,V,,,,,,,110523,,,N*63
22:41:45.051 -> $GNGGA,214146.00,,,,,0,00,99.99,,,,,,*7C
22:41:45.139 -> $GNRMC,214146.10,V,,,,,,,110523,,,N*62
22:41:45.184 -> $GNGGA,214146.10,,,,,0,00,99.99,,,,,,*7D
22:41:45.243 -> $GNRMC,214146.20,V,,,,,,,110523,,,N*61
22:41:45.243 -> $GNGGA,214146.20,,,,,0,00,99.99,,,,,,*7E
22:41:45.337 -> $GNRMC,214146.30,V,,,,,,,110523,,,N*60
22:41:45.391 -> $GNGGA,214146.30,,,,,0,00,99.99,,,,,,*7F
22:41:45.437 -> $GNRMC,214146.40,V,,,,,,,110523,,,N*67
22:41:45.437 -> $GNGGA,214146.40,,,,,0,00,99.99,,,,,,*78
22:41:45.578 -> $GNRMC,214146.50,V,,,,,,,110523,,,N*66
22:41:45.578 -> $GNGGA,214146.50,,,,,0,00,99.99,,,,,,*79
22:41:45.625 -> $GNRMC,214146.60,V,,,,,,,110523,,,N*65
22:41:45.625 -> $GNGGA,214146.60,,,,,0,00,99.99,,,,,,*7A
```

mpvlad, 2023-05-15:
@PaulZC Thank you very much for your suggestion! That was exactly what was going on! The code now is working as expected :)