sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #201: Negative elevation in NAVSAT data?

nicolasff, 2023-08-06:
### Subject of the issue

Hello,

I'm using this library with a [SparkFun NEO-M9N GPS breakout](https://www.sparkfun.com/products/15712), and while it's been working well so far there's still something I don't understand about the NAVSAT data.

I'm using `setAutoNAVSATcallbackPtr` to get callbacks containing a `UBX_NAV_SAT_data_t *` value, from which I can get an array of `UBX_NAV_SAT_block_t` structs. The first few fields give some info about the vehicle, including its elevation and azimuth: https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/8a33fb93e92a6ab152097f143bcbc84c7ad05504/src/u-blox_structs.h#L1127-L1134

Azimuth being in degrees from 0 to 360 makes sense to me, but the range for elevation does not: it goes from -90 to +90. How can a satellite have negative elevation? This would mean it's below the horizon.

In addition, I _am_ actually seeing negative values in the NAVSAT callback, but with `elev` = -91, even outside of the stated range.
Here are some values from a recent read:

```
gnssID: 0, azim: 212, elev: 31
gnssID: 0, azim: 202, elev: 19
gnssID: 0, azim: 320, elev: 76
gnssID: 0, azim: 293, elev: 47
gnssID: 0, azim: 326, elev: 8
gnssID: 0, azim: 279, elev: 19
gnssID: 0, azim: 139, elev: 22
gnssID: 0, azim: 197, elev: 16
gnssID: 0, azim: 104, elev: 36
gnssID: 0, azim: 46, elev: 24
gnssID: 0, azim: 51, elev: 54
gnssID: 0, azim: 82, elev: 3
gnssID: 1, azim: 171, elev: 46
gnssID: 1, azim: 191, elev: 46
gnssID: 1, azim: 156, elev: 43
gnssID: 2, azim: 0, elev: -91       <---------
gnssID: 3, azim: 0, elev: -91       <---------
gnssID: 6, azim: 90, elev: 27
```

Note the two towards the end with `azim: 0` and `elev: -91`. I don't think I've seen this -91 elevation without the azimuth also being zero.

So my questions are:
1. What does it mean for the elevation to be negative and how should I understand this data in terms of where the vehicle is, relative to the receiver?
2. What are these `-91` values for elevation? What should I do about them?

### Your workbench
* What development board or microcontroller are you using? → ESP32-S3
* What version of hardware or breakout board are you using? → NEO-M9N, mentioned above
* How is the breakout board wired to your microcontroller? → I²C via Qwiic port
* How is everything being powered? → 5V USB for the MCU, 5V for the power wire of the Qwiic cable
* Are there any additional details that may help us help you?

### Steps to reproduce
```cpp
static void neo_m9n_navsat_callback(UBX_NAV_SAT_data_t *sat) {
    for (int i = 0; i < sat->header.numSvs; i++) {
        Serial.printf("    #%2d -- [%s]"
                " GNSS ID: %" PRIu8 ", SV ID: %3" PRIu8 " (%s)"
                ", CNO_ratio: %2" PRIu8 ", elev: %3" PRId8 " deg, azim: %3" PRId16 " deg"
                "\n",
                i, sat->blocks[i].flags.bits.svUsed ? "Y" : "N",
                sat->blocks[i].gnssId, sat->blocks[i].svId, gnssName,
                sat->blocks[i].cno, sat->blocks[i].elev, sat->blocks[i].azim);
    }
}

static SFE_UBLOX_GNSS ublox;

void setup() {
    Wire.begin(I2C_SDA, I2C_SCL);
    ublox.begin(Wire, 0x42);

    ublox.setI2COutput(COM_TYPE_UBX);
    ublox.setAutoNAVSATcallbackPtr(neo_m9n_navsat_callback);
    ublox.setNavigationFrequency(2); // 2 solutions/sec
}

void loop() {
    ublox.checkUblox();
    ublox.checkCallbacks(); 
}
```

### Expected behavior
Satellites should be above the horizon.

### Actual behavior
Some satellites appear to be almost exactly on the opposite side of the Earth.

PaulZC, 2023-08-06:
Hi Nicolas (@nicolasff ),

I think an elevation of -91 (and an azimuth of 0) simply means the values are "unknown". The module is aware of that satellite but it is not currently in view (from your position). When in view, the elevation should always be positive.

Background:

NAV SAT contains the same azimuth and elevation as the NMEA GnGSV message:

[image]

I think the correct term for the coordinate system is "Topocentric-Horizon".

The elevation +/- 90 comment in the code comes straight from the u-blox interface description: 

```
elev; deg; Elevation (range: +/-90), unknown if out of range
```

The comment in the GSV description makes more sense:

```
elv; numeric; deg; Elevation (<= 90)
```

Discard any SVs with an elevation of -91 and away you go...!

Best wishes,
Paul

closed 2023-08-06 by PaulZC

PaulZC, 2023-08-06:
Actually... Thinking about this a bit more. I think the SV elevation could be (slightly) negative - particularly if you're airborne. But -91 is clearly not right and must - I think - be the value for "unknown".

Best wishes,
Paul

nicolasff, 2023-08-07:
Thanks Paul!

I hadn't thought about an airborne observer, it does explain how negative values could be seen.
The comment saying "unknown if out of range" is pretty explicit, and the value -91 being the first that's out of the stated range does seem to indicate that it was specifically chosen to represent this "unknown" case. I had been discarding those, but it's good to have confirmation.

Best,
Nicolas