sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #202: Fails to initialise

StressedDave, 2023-09-10:
### Subject of the issue

myGNSS.begin() returns false

### Your workbench

Using Arduino v2.2.1 & CLI 0.34.0

Behaviour has been confirmed on both Sparkfun NEO-M8U & NEO-M9N boards
Behaviour confirmed with Adafruit Feather boards (M0, M4 and RP2040) & QtPy (ESP32 C3)

Device (Adafruit Feather STM32F403) using older version of library (last compiled several months ago) works correctly with both GPS boards

Device scan confirms that boards are connected

### Steps to reproduce

Tested on large number of library examples, including Example8_GetProtocolVersion

### Expected behavior

Expected to work as standard example

### Actual behavior

Device outputs "u-blox GNSS not detected at default I2C address. Please check wiring. Freezing." via serial port. Debug output shows (x3):
```
Sending: CLS:CFG ID:0x0 Len: 0x1 Payload: 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 100 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 56 bytes
checkUbloxI2C: OK, zero bytes available
checkUbloxI2C: Reading 28 bytes
checkUbloxI2C: OK, zero bytes available
waitForACKResponse: TIMEOUT after 1100 msec.
```

PaulZC, 2023-09-10:
Hi,

I can't replicate your issue...

I've just tried Example8 on an Adafruit SAMD21 M0 (Adalogger) connected to a NEO-M8U breakout via I2C and everything is working fine.

I'm using: Arduino IDE 1.8.19; Adafruit SAMD boards 1.7.13; this library v2.2.24.

Are you sure you are using _**this**_ library - v2.2.24? Or are you perhaps trying to use [v3 of the library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3)? (Your comment about "using older version of library" hints that you may be using v3?)

Best,
Paul

PaulZC, 2023-09-10:
Humm... From the "Sending: CLS:CFG ID:0x0", it looks like you are indeed using this library. That is the CFG-PRT check used by v2 of the library, not v3.

Just tried Arduino IDE 2.2.1 and again it's working fine for me:

[image]

All I can suggest is that you check your I2C cable and make sure you are connecting to the correct SCL / SDA pins?

Closing - because I can't replicate your issue. Please re-open if you think I've missed something.

Best wishes,
Paul

closed 2023-09-10 by PaulZC

StressedDave, 2023-09-10:
Weird - I didn't have the v3 library on my system. A quick install of v3, deinstall of v3 & v2 and then reinstall v2 and it's happy now. 

I'm blaming the Arduino library manager...

Thanks for checking for me.

PaulZC, 2023-09-10:
No worries - glad it's working for you,
Paul