sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #203: [Feature Request] Controlled shutdown for Hot Start.

Eric-FR, 2023-09-22:
### Subject of the issue
In order to take advantage of Hot Start with F9P and F9R (and maybe some others chips), one have to perform a controlled GNSS stop with BBR storage of parameters before power off. Hot Start may be very valuable (time saving) with F9R by avoiding or reducing IMU calibration and alignment process. This must be done with the UBX-CFG-RST command. Can this be added to the library?

### Suggestion
```
void SFE_UBLOX_GNSS::controlledGNSSStop()
{
  // Issue controlled GNSS stop
  // Necessary for Hot Start after power off
  packetCfg.cls = UBX_CLASS_CFG;
  packetCfg.id = UBX_CFG_RST;
  packetCfg.len = 4;
  packetCfg.startingSpot = 0;
  payloadCfg[0] = 0;          // hot start
  payloadCfg[1] = 0;          // hot start
  payloadCfg[2] = 0x08;       // 0x08 = Controlled GNSS stop
  payloadCfg[3] = 0;          // reserved
  sendCommand(&packetCfg, 0); // don't expect ACK
}

```

In opposite, I would also like to force Cold Start like:

```
void SFE_UBLOX_GNSS::forcedColdStart()
{
  // Issue Controlled software reset with BBR clearing
  packetCfg.cls = UBX_CLASS_CFG;
  packetCfg.id = UBX_CFG_RST;
  packetCfg.len = 4;
  packetCfg.startingSpot = 0;
  payloadCfg[0] = 0xFF;        // cold start
  payloadCfg[1] = 0xFF;        // cold start
  payloadCfg[2] = 0x01;       // Controlled software reset
  payloadCfg[3] = 0;            // reserved
  sendCommand(&packetCfg, 0); // don't expect ACK
}

(And, overall, is there a general process to trigger a command like UBX-CFG-RST with user parameters and not defining a function for each case?)

cturvey, 2023-09-22:
Commenting to loop myself into the conversation.
https://forum.sparkfun.com/viewtopic.php?f=116&t=60251
There's also UBX-ESF-RESETALG to reset the fusion
https://portal.u-blox.com/s/question/0D52p0000DyEgTQCQ0/f9r-imu-alignment

PaulZC, 2023-09-23:
Hi Eric (@Eric-FR ),

For the ZED-F9R, I'd recommend using [version 3 of this library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3). It configures the module using the Configuration Interface keys instead of the deprecated UBX-CFG-PRT and UBX-CFG-MSG messages.

I think all of the functions you request are already available in v3. Please see:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/blob/05eea9878f1e91575781d86b82642847fccce2cf/src/u-blox_GNSS.h#L365-L367

If something is missing, please open a Feature Request issue in v3. Thank you.

Best wishes,
Paul

closed 2023-09-23 by PaulZC

PaulZC, 2023-09-23:
I've opened an issue for ESF-RESETALG:

https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/issues/35

Eric-FR, 2023-09-23:
I've switched to V2 a couple of months ago. I just discovered the V3 this week while checking libraries in Arduino IDE. Will have to switch to it faster than expected.

Eric-FR, 2023-09-23:
BTW, I made the switch to the V3 and recompilation of my code went fine.