sparkfun/SparkFun_u-blox_GNSS_Arduino_Library #205: The program stops midway.

drinkBr, 2023-10-09:
I have been working on this problem for over two months. I was running "Example1_GetPositionAccuracy.ino" on Arduino UNO. However, without any warning, it suddenly became unable to execute after a certain point. I would like to receive advice on how to solve this issue. I will provide information about the device I am using, as well as the results from Arduino IDE and the serial monitor.
・Arduino UNO
・SparkFun Qwiic Shield for Arduino
・GPS-15192
・GPS-16481
[image]
[image]

Thanks.

PaulZC, 2023-10-09:
Hi,

Thank you for reporting this. I have a few comments and questions:

For the ZED-F9P (GPS-16481) we recommend [Version 3 of this library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3). But I do not think that is your issue.

The Arduino UNO only has a very small amount of RAM (2kBytes). That could be your issue. But it does not explain why is was working and now is not working.

Please copy and paste the text from the Serial Monitor, then we can translate it.

Have you made any changes to Example1?

How are you powering the Arduino UNO?

You say "the program stops midway". Does it stop at the ```begin``` with the error ```u-blox GNSS not detected at default I2C address. Please check wiring. Freezing.```? Or does it stop in the ```loop``` when printing the ```Lat:``` etc.?

Thank you,
Paul

drinkBr, 2023-10-10:
Hey, Paul
> For the ZED-F9P (GPS-16481) we recommend [Version 3 of this library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3). But I do not think that is your issue.

Thanks. I will give it a try.

> Please copy and paste the text from the Serial Monitor, then we can translate it.

The text on the serial monitor is as follows.

SparkFun u-blox Example

It only displays up to this point. Also, the output is displayed as follows.

最大32256バイトのフラッシュメモリのうち、スケッチが30016バイト（93%）を使っています。
最大2048バイトのRAMのうち、グローバル変数が917バイト（44%）を使っていて、ローカル変数で1131バイト使うことができます。

> Have you made any changes to Example1?

No.

> How are you powering the Arduino UNO?

It is powered from the computer's USB port.

> You say "the program stops midway". Does it stop at the begin with the error u-blox GNSS not detected at default I2C address. Please check wiring. Freezing.? Or does it stop in the loop when printing the Lat: etc.?

Upon checking, it seems that the program is not running from line 40 (as follows).

  if (myGNSS.begin() == false) //Connect to the u-blox module using Wire port
  {
    Serial.println(F("u-blox GNSS not detected at default I2C address. Please check wiring. Freezing."));
    while (1);
  }

Thanks.

PaulZC, 2023-10-10:
Hi,

The message is about the memory use. It translates to:

```
Out of the maximum 32,256 bytes of flash memory, the sketch uses 30,016 bytes (93%).
Of the maximum 2048 bytes of RAM, global variables use 917 bytes (44%), and local variables can use 1131 bytes.
```

I wonder if the GNSS module needs to be reset? Do you know how to use [u-blox u-center](https://www.u-blox.com/en/product/u-center)? You can reset the module by connecting to the GNSS USB port and using the UBX-CFG-CFG message:

Connect the board to your PC via USB and open up u-center.
Select the correct COM port and connect to the module.
You should see the satellite signal strength 'bar chart' etc. appear?
Open View \ Messages View
Select UBX-CFG-CFG in the message tree
Select "Revert to default configuration" and click Send at the bottom of the window
Then click "Save current configuration" and click Send at the bottom of the window

I hope this helps.

Best wishes,
Paul

PaulZC, 2023-10-10:
Hi,

I have tested Example1 on: Arduino UNO R3 (ATMEGA328P); SparkFun RedBoard (ATMEGA328P); Arduino UNO R4 WiFi (Renesas RA41M1). It works correctly on all three boards. I can not replicate your issue. Please do try resetting the ZED-F9P using u-center. Please also check the soldering on the Qwiic Shield. Perhaps you have a bad solder joint?

Best wishes,
Paul

drinkBr, 2023-10-17:
Hi, Paul
I'm sorry for the late reply.

> I have tested Example1 on: Arduino UNO R3 (ATMEGA328P); SparkFun RedBoard (ATMEGA328P); Arduino UNO R4 WiFi (Renesas RA41M1). It works correctly on all three boards.

When I tried it with Arduino Uno R3, it worked correctly. There were no issues with version 3 of this library. Thanks.

I have a couple of questions.

First question: What does "3D Positional Accuracy" in the program refer to?
Second question: In the Example1 program, I believe the ZED-F9P is used as a GPS. However, if I want to perform relative positioning through a reference station in a network environment, could you please let me know if there is a sample program available for this?

Thanks.

PaulZC, 2023-10-17:
Hi,

1) The position accuracy is ```pAcc``` (Position Accuracy Estimate) from the UBX-NAV-HPPOSECEF message. It is statistical. Please see: https://forum.sparkfun.com/viewtopic.php?p=243464#p243464

2) It depends what you mean by "relative positioning". You can use two ZED-F9Ps to provide relative positioning (RELPOSNED) in a "moving base" configuration: https://www.u-blox.com/sites/default/files/documents/ZED-F9P-MovingBase_AppNote_UBX-19009093.pdf . Or you could use an NTRIP Caster to share RTCM correction data from a fixed Base to a Rover. Our Reference Station product allows the Base and Rover to be connected via Ethernet : https://www.sparkfun.com/products/22429 .

Best wishes,
Paul

closed 2023-10-17 by PaulZC